dorssel/usbipd-win #962: how can I have network acess with no translation?

skunx99, 2024-05-25:
`ulao@entertainer:/mnt/c/Users/Administrator/AppData/Local/Microsoft/WindowsApps$ ping google.com
PING google.com (172.217.165.206) 56(84) bytes of data.
64 bytes from lax31s06-in-f14.1e100.net (172.217.165.206): icmp_seq=1 ttl=118 time=20.2 ms
64 bytes from lax31s06-in-f14.1e100.net (172.217.165.206): icmp_seq=2 ttl=118 time=26.0 ms
64 bytes from lax31s06-in-f14.1e100.net (172.217.165.206): icmp_seq=3 ttl=118 time=19.5 ms
`

but
`
C:\Windows\System32>netsh interface show interface

Admin State    State          Type             Interface Name
-------------------------------------------------------------------------
Enabled        Disconnected   Dedicated        Ethernet 3
Enabled        Disconnected   Dedicated        Wi-Fi
Enabled        Connected      Dedicated        Local Area Connection
`

I learned about this trying.

`
C:\Windows\System32>usbipd wsl attach --busid 2-3
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
usbipd: error: The local IP address for the WSL virtual switch could not be found.
`

Other details is this. 
`
C:\Windows\System32> wsl --set-version Ubuntu-22.04 2
For information on key differences with WSL 2 please visit https://aka.ms/wsl2
Conversion in progress, this may take a few minutes.
The distribution is already the requested version.
Error code: Wsl/Service/WSL_E_VM_MODE_INVALID_STATE
`

dorssel, 2024-05-26:
> usbipd wsl attach --busid 2-3

That's an old version of `usbipd-win`. What does `usbipd --version` say? You probably should update to the latest release.

skunx99, 2024-05-26:
yeah looks like what it was

dorssel, 2024-05-27:
Did you have any devices "forced bound"? That does sometimes block updates and indeed requires uninstall/re-install.

Does it work now?

skunx99, 2024-05-27:
yeah, after removing the old and reinstalling the new its back up and running.

closed 2024-05-28 by dorssel