dorssel/usbipd-win #964: Permissions - solved

dzid26, 2024-05-26:
After device was visible in `lsusb` (wsl ubuntu 20.04), I was struggling to do anything else. 
```
lsusb -v 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Couldn't open device, some information will be missing
```
More details of what I tried:
https://gitlab.com/alelec/wsl-usb-gui/-/issues/27

I also tried `sudo usermod -a -G plugdev {user}`

Last commands that finally made it work was:
` sudo chmod -R 777 /dev/bus/usb/`

dorssel, 2024-05-26:
You solved an issue reported in another project, and now you *open* an issue here. What am I missing?

dzid26, 2024-05-26:
Sorry for confusion.
I just wanted to leave a trace for others of a possible solution.

closed 2024-05-26 by dzid26