dorssel/usbipd-win #969: usb device suddenly works. But why?

comsyspro, 2024-06-04:
Here is a USB device that has about 7 USB devices in it. This USB device was causing a few problems and I would like to ask if there have been any changes to the code recently. I compiled usbipd manually with the current code and the device now works. I also compiled the current code on the client side and it could also be due to the client code that this USB device suddenly works. Can you say whether you have made any changes to the code recently to better support such USB devices?

[image]

dorssel, 2024-06-05:
No, nothing substantial has changed from 4.2.0 release in the current master branch. I think it was the client side changes. Can you try again with a 4.2.0 release?

comsyspro, 2024-06-06:
Thank you. Indeed the changes are on the client side. I went back to 4.2.0 and it worked also. On client side I also went back to previous version and the problems started again. So I'm sure now that the fixes ar done on the client side.

closed 2024-06-06 by comsyspro