dorssel/usbipd-win #973: Using 5.15.xx kernel not equipped

573, 2024-06-12:
```
[wsl2]
memory=8GB   # Limits VM memory in WSL 2 up to 3GB
processors=2 # Makes the WSL 2 VM use two virtual processors
```

```
$ uname -sr
Linux 5.15.146.1-microsoft-standard-WSL2
```

admin powershell:
```
usbipd bind --busid 1-2
usbipd: info: Device with busid '1-2' was already shared.
```

non-admin powershell:
```
usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-2    04a9:220e  CanoScan                                                      Shared

usbipd attach --wsl --busid 1-2
usbipd: info: Using WSL distribution 'NixOS-Groundtruth' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.
```

573, 2024-06-12:
# Steps using [Troubleshooting](https://github.com/dorssel/usbipd-win/wiki/Troubleshooting) so far

```
# non-admin pwsh
winget uninstall usbipd-win
winget install --interactive --exact dorssel.usbipd-win -v 3.1.0   # this worked so far but has firewall rule warning
usbipd bind --busid 1-2
usbipd wsl attach -a --busid 1-2
usbipd: info: Using default WSL distribution 'NixOS-Groundtruth'; specify the '--distribution' option to select a different one.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.

# admin pwsh
sc.exe stop usbipd
 usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      3.1.0+66.Branch.master.Sha.47e0500785e1ad4bd6574da02dae33744ea23f52
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 127.0.0.1
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 10 ms
info: Usbipd.ConnectedClient[1]
      Client 127.0.0.1 claimed device at 1-2 (USB\VID_04A9&PID_220E\5&665DA22&0&2).
dbug: Usbipd.AttachedClient[1000]
      Masked USB_CONFIG_REMOTE_WAKEUP
dbug: Usbipd.AttachedClient[1000]
      Masked USB_CONFIG_REMOTE_WAKEUP
dbug: Usbipd.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1

# non-admin pwsh (usbipd wsl attach -a --busid 1-2 still running)
usbip: error: tcp connect
usbip: error: Attach Request for 1-2 failed - Device in error state
usbip: error: Attach Request for 1-2 failed - Device not found
Attached
Detached
usbip: error: tcp connect
Attached

# again in wsl now
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 04a9:220e Canon, Inc. CanoScan N1240U/LiDE 30
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

dorssel, 2024-06-12:
> usbipd: info: Using WSL distribution 'NixOS-Groundtruth' to attach; the device will be available in all WSL 2 distributions.
> usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.

I'm sure your kernel is WSL capable. It's just the test for it that fails due to an uncommon distro. To solve it, you could simply install another distro (e.g. Ubuntu) and use that to attach. The device will be available in every distro.

However, I would like to find out why the test fails on your distro. This is the command that fails:

```
wsl --distribution NixOS-Groundtruth --user root --cd / --exec cat /sys/devices/platform/vhci_hcd.0/status
```

What is the output if you run that in a (Windows) command prompt?

573, 2024-06-13:
@dorssel `wsl --distribution NixOS-Groundtruth --user root --cd / --exec /run/current-system/sw/bin/cat /sys/devices/platform/vhci_hcd.0/status`
>hub port sta spd dev      sockfd local_busid
hs  0000 004 000 00000000 000000 0-0
hs  0001 004 000 00000000 000000 0-0
hs  0002 004 000 00000000 000000 0-0
hs  0003 004 000 00000000 000000 0-0
hs  0004 004 000 00000000 000000 0-0
hs  0005 004 000 00000000 000000 0-0
hs  0006 004 000 00000000 000000 0-0
hs  0007 004 000 00000000 000000 0-0
ss  0008 004 000 00000000 000000 0-0
ss  0009 004 000 00000000 000000 0-0
ss  0010 004 000 00000000 000000 0-0
ss  0011 004 000 00000000 000000 0-0
ss  0012 004 000 00000000 000000 0-0
ss  0013 004 000 00000000 000000 0-0
ss  0014 004 000 00000000 000000 0-0
ss  0015 004 000 00000000 000000 0-0

You're right the problem might be `cat` is not on the path by default in my distro, unmodified `wsl --distribution NixOS-Groundtruth --user root --cd / --exec cat /sys/devices/platform/vhci_hcd.0/status` gives:
><3>WSL (2897) ERROR: CreateProcessEntryCommon:505: execvpe cat failed 2
<3>WSL (2897) ERROR: CreateProcessEntryCommon:508: Create process not expected to return

dorssel, 2024-06-14:
Does the following work?

```
wsl --distribution wsl-usbip --user root --cd / --exec sh -c "cat /sys/devices/platform/vhci_hcd.0/status"
```

and/or this one?

```
wsl --distribution wsl-usbip --user root --cd / --exec /bin/sh -c "cat /sys/devices/platform/vhci_hcd.0/status"
```

Pheoxy, 2024-06-14:
> Does the following work?
> 
> ```
> wsl --distribution NixOS-Groundtruth --user root --cd / --exec sh -c "cat /sys/devices/platform/vhci_hcd.0/status"
> ```
> 
> and/or this one?
> 
> ```
> wsl --distribution NixOS-Groundtruth --user root --cd / --exec /bin/sh -c "cat /sys/devices/platform/vhci_hcd.0/status"
> ```

To further this as I am using NixOS as well and having the same problem, here is the results of each command.

```
$ wsl --distribution NixOS --user root --cd / --exec sh -c "cat /sys/devices/platform/vhci_hcd.0/status"
/nix/store/rdd4pnr4x9rqc9wgbibhngv217w2xvxl-bash-interactive-5.2p26/bin/sh: line 1: cat: command not found
```
```
$ wsl --distribution NixOS --user root --cd / --exec /bin/sh -c "cat /sys/devices/platform/vhci_hcd.0/status"
/nix/store/rdd4pnr4x9rqc9wgbibhngv217w2xvxl-bash-interactive-5.2p26/bin/sh: line 1: cat: command not found
```

dorssel, 2024-06-15:
If NixOS cannot even run Linux Standard Base Core applications, then it is not going to be supported. Use another distro to attach.

Pheoxy, 2024-06-16:
> If NixOS cannot even run Linux Standard Base Core applications, then it is not going to be supported. Use another distro to attach.

As this seems to be a issue with how NixOS uses packages I normally get around it by using paths like `/usr/bin/env bash` in my scripts so it still works within other distros, it being a problem with usbipd not finding `cat` may mean other commands might be unavailable as well.

I agree on not going out of your way to support this, in fact I'm more of a scripter than programmer so I couldn't way in on the complexity of supporting NixOS but don't waste your time.

I'm sure that theirs probably a NixOS option that can be used to allow support and as long as the standard distro support works there may be a NixOS specific option we can enable when using usbipd like we do with vscode.

dorssel, 2024-06-16:
One option would be (if NixOS somehow is capable) to ensure the required commands are in the default path; on regular OSs this is in `/etc/environment`. The commands used by `usbipd-win` are:
- `cat`
- `/bin/sh`
- `/bin/wslinfo` (injected by WSL, should be available always)

Wikipedia explicitly says: 

> An implication of this is that NixOS does not follow the [Filesystem Hierarchy Standard](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard).

But like I said, the easiest is probably to just install Ubuntu as well, and use that for attaching. The device should then be available in NixOS as well.

saippua, 2024-08-10:
I believe this issue is related to https://github.com/nix-community/NixOS-WSL/issues/284#issuecomment-1712478703

For me the line that fails is in `Wsl.cs` the bit where it tries to read the kernel version using a command like:
```
wsl --distribution NixOS --user root --cd / --exec cat /proc/config.gz
```
I've confirmed this produces an error when executed in PS.
the following works (i.e. same without `--exec`):
```
wsl --distribution NixOS --user root --cd / cat /proc/config.gz
```

Booting up another distro and attaching through that works, however I'm not super thrilled about having a random ubuntu stub in my configurations, and it's also tedious to boot the stub every time I need to reattach a device.

is there any chance we could use a login shell to read the kernel version, or perhaps fallback to one if a non-login shell fails?

I'd be interested in fiddling around with this a bit more in a fork, but I haven't been able to find any build instructions.

dorssel, 2024-08-10:
@saippua 
The login shell doesn't have to be a `sh`-type shell. Could be PowerShell, which has a different syntax (not for `cat`, but for some of the other commands).

Does the following work?

```
wsl --distribution NixOS --user root --cd / --exec sh -c 'cat /proc/config.gz'
```

and this one?

```
wsl --distribution NixOS --user root --cd / --exec /bin/sh -c 'cat /proc/config.gz'
```

If we are going the "fallback" route (I like the idea), then we might as well go all the way:

1) --exec cat
2) --exec sh -c cat (with or without prefix /bin)
3) cat (using whatever login shell happens to be configured)

And this for all the commands, of course.

dorssel, 2024-08-14:
From https://pubs.opengroup.org/onlinepubs/9799919799/utilities/sh.html, the `-l` option isn't actually standardized. And the `/bin` prefix assumes FHS, which if it is  available isn't actually necessary (so we can skip that version). Therefore, I will make any command `xxx` use the following fallback sequence:

1) `--exec xxx` (best for performance, if it works)
2) `--exec sh -c xxx` (POSIX)
3) `--exec sh -lc xxx` (for non-FHS, such as NixOS)
4) `xxx` (using whatever default shell is configured)

qwqawawow, 2024-11-16:
Sorry to bother Here 
Small hack for other 
```
sudo ln -s /run/current-system/sw/bin/cat /usr/bin
sudo ln -s /run/current-system/sw/bin/usbip /usr/bin
```
From https://github.com/nix-community/NixOS-WSL/issues/111#issuecomment-2027966060

danylo-volchenko, 2025-07-08:
Hello!
Facing the same issue with NixOS and custom kernel to support usbip.
```
PS C:\Users\111> usbipd attach --wsl --busid 7-1
usbipd: info: Using WSL distribution 'NixOS' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.
```

Here's the info and some checks from previous comments on that one:
```
~  :: =>> uname -a
Linux nixos 6.6.87.2-microsoft-standard-WSL2+ #8 SMP PREEMPT_DYNAMIC Tue Jul  8 10:02:06 EEST 2025 x86_64 GNU/Linux

~  :: =>> grep -i 'usbip' WSL2-Linux-Kernel/.config
CONFIG_USBIP_CORE=y
CONFIG_USBIP_VHCI_HCD=y
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
CONFIG_USBIP_HOST=y
# CONFIG_USBIP_DEBUG is not set

~  :: =>> cat /etc/wsl.conf
[automount]
enabled=true
ldconfig=false
mountFsTab=false
options=metadata,uid=1000,gid=100
root=/mnt

[boot]
systemd=true

[interop]
appendWindowsPath=true
enabled=true

[network]
generateHosts=true
generateResolvConf=true
hostname=nixos

[user]
default=wired

~  :: =>> dmesg | grep -i usb
[    0.113290] ACPI: bus type USB registered
[    0.113290] usbcore: registered new interface driver usbfs
[    0.113290] usbcore: registered new interface driver hub
[    0.113290] usbcore: registered new device driver usb
[    0.228252] usbcore: registered new interface driver cdc_acm
[    0.228512] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    0.228798] usbcore: registered new interface driver cdc_wdm
[    0.229017] usbcore: registered new interface driver usbserial_generic
[    0.229228] usbserial: USB Serial support registered for generic
[    0.229687] vhci_hcd vhci_hcd.0: USB/IP Virtual Host Controller
[    0.230085] vhci_hcd vhci_hcd.0: new USB bus registered, assigned bus number 1
[    0.230788] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.06
[    0.231116] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.231446] usb usb1: Product: USB/IP Virtual Host Controller
[    0.231733] usb usb1: Manufacturer: Linux 6.6.87.2-microsoft-standard-WSL2+ vhci_hcd
[    0.232088] usb usb1: SerialNumber: vhci_hcd.0
[    0.232592] hub 1-0:1.0: USB hub found
[    0.233509] vhci_hcd vhci_hcd.0: USB/IP Virtual Host Controller
[    0.233886] vhci_hcd vhci_hcd.0: new USB bus registered, assigned bus number 2
[    0.234275] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    0.234902] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.06
[    0.235241] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.235521] usb usb2: Product: USB/IP Virtual Host Controller
[    0.235751] usb usb2: Manufacturer: Linux 6.6.87.2-microsoft-standard-WSL2+ vhci_hcd
[    0.236023] usb usb2: SerialNumber: vhci_hcd.0
[    0.236394] hub 2-0:1.0: USB hub found
[    0.237310] usbcore: registered new device driver usbip-host
```

```
PS C:\Users\111> wsl -v
WSL version: 2.5.9.0
Kernel version: 6.6.87.2-1
WSLg version: 1.0.66
MSRDC version: 1.2.6074
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
Windows version: 10.0.26100.4349
```

did:
```
sudo ln -s /run/current-system/sw/bin/cat /usr/bin
sudo ln -s /run/current-system/sw/bin/usbip /usr/bin
```

`wsl --distribution NixOS --user root --cd / --exec sh -c 'cat /proc/config.gz'` - works
`wsl --distribution NixOS --user root --cd / --exec /bin/sh -c 'cat /proc/config.gz'` - works

```
PS C:\Users\111> wsl --distribution NixOS --user root --cd / --exec cat "/sys/devices/platform/vhci_hcd.0/status"
hub port sta spd dev      sockfd local_busid
hs  0000 004 000 00000000 000000 0-0
hs  0001 004 000 00000000 000000 0-0
hs  0002 004 000 00000000 000000 0-0
hs  0003 004 000 00000000 000000 0-0
hs  0004 004 000 00000000 000000 0-0
hs  0005 004 000 00000000 000000 0-0
hs  0006 004 000 00000000 000000 0-0
hs  0007 004 000 00000000 000000 0-0
ss  0008 004 000 00000000 000000 0-0
ss  0009 004 000 00000000 000000 0-0
ss  0010 004 000 00000000 000000 0-0
ss  0011 004 000 00000000 000000 0-0
ss  0012 004 000 00000000 000000 0-0
ss  0013 004 000 00000000 000000 0-0
ss  0014 004 000 00000000 000000 0-0
ss  0015 004 000 00000000 000000 0-0
```

dorssel, 2025-07-08:
This has been fixed in the current master build. You can install that, or wait for the next release (soon TM).

closed 2025-08-15 by dorssel