dorssel/usbipd-win #974: Location is not available | Access is denied| Can't read flash drives

Katashynskyi, 2024-06-15:
After attempt to connect to my "radiomaster pocket" joystick to wsl2 via usbipd i tried to connect my another USB devices, like external hard drives. So after usage of command like "bind" and "force bind" and "auto-attach device" I face "access is denied" problem on my windows 11 side. I'd like to know what exactly usbipd-win change at my windows part so I can use flash drives and hard drives again.
Also, i try wsl-usb-gui with the same result.
Please help)
[image]
[image]

dorssel, 2024-06-15:
With `usbipd bind --force` you have changed the driver permanently, so Windows cannot use the device any longer. To undo everything, simply use `usbipd unbind --all`, or uninstall the software.

Katashynskyi, 2024-06-18:
Hi dorssel. Sorry but it didn't work. 
I tried: (to connect radiomaster pocket joystick)
Probably in this order.
install usbipd on wsl2 side
build WSL 2 kernel with additional drivers
install GUI usbipd (by Andrew Leech)
joystick seen by wsl2 commands as a device but not working... 

Then i tried use external harddrive instead (mostly via GUI usbipd)
then Access denied error occured
tried to delete everything:
uninstall usbipd GUI (which by the way can be the reason of my problem), (there's no unbind --all command or force unbind, but force bind is),
uninstall usbipd, 
uninstall wsl2 (ubuntu 22.04)
Then tried to install usbipd again and still "access is denied.
So where can be a mistake here?
(Also i tried to fix "USB Access Denied" error by it's own, it didn't help either)

Katashynskyi, 2024-06-19:
I just figured it out. Delete usbipd. Then go to Device Manager -> Universal Serial Bus controllers -> uninstall device. Also, uninstall all of "other devices."
Thsnks for help.

dorssel, 2024-06-22:
You seem to address multiple, isolated, issues.

1) What is the problem with the first device (joystick).
2) What are you trying to do with the flash drive?

What commands are you giving? And what is the output? Please discuss one problem at a time.

Katashynskyi, 2024-06-23:
Main issue I'm referring to is my access denied for any flash drives I'm trying to plug into any USB on my host Windows PC.
1. I couldn't connect it. 
2. I'm trying to open it on host pc.
I've tried "usbipd bind" and "usbipd unbind --all", "usbipd attach" and "usbipd detach". And I still can't access to any flash drives which I connect to host pc.

dorssel, 2024-06-23:
If you want to use a flash drive on your host PC, why exactly do you want to use `usbipd-win`? That is to access USB devices on guest PCs...

What do you mean with 'connect it'?

Katashynskyi, 2024-06-23:
At first, I've tried to connect my joystick to the wsl2 using usbipd-win, but I failed, then I've tried to connect other devices (flash drives) and failed too. As a result, when I plug in (connect) any flash drive via USB I'm getting Access is denied error. 

So, I can't use flash drives on either the host machine or the guest.

Maybe you can suggest what I should fix on the Windows side first (since deleting usbipd didn't help).

I've tried to delete WSL2, but it didn't fix the problem. However, if I run Windows in "Safe Mode" with usbipd deleted beforehand, everything is okay, and I have access.

dorssel, 2024-06-23:
My guess is that the "owner" of the drive's root directory has changed. As an admin you can fix that by going to the security settings of the drive and change the access rights. However, there really is *no way* that `usbipd-win` has caused this. It does not interact with the device at the application level at all. This really is not a `usbipd-win` issue.

dorssel, 2025-02-02:
Stale

closed 2025-02-02 by dorssel