dorssel/usbipd-win #975: Unable to "attach" usb device through WSL

NickM1111, 2024-06-19:
Hi,
Similar to other posts I have seen, I am having difficulty attaching my usb device to the WSL.
The error message I am getting is:
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues. 

I am running the latest version of usbipd-win_4.2.0.

Can you please advise on how to fix?
Thanks

dorssel, 2024-06-24:
What is the version of Ubuntu?
After you've tried to attach, what is the output of `mount` in WSL?

shufang-ding, 2024-07-16:
> What version of Ubuntu is it? What is the output from `mount`WSL after trying to connect?

Yes, I also encountered this problem：

Persisted:
GUID                                  DEVICE

PS C:\Windows\system32> usbipd bind --busid 3-4
PS C:\Windows\system32> usbipd attach --wsl  --busid 3-4
usbipd: error: Windows Subsystem for Linux version 2 is not available. See https://learn.microsoft.com/windows/wsl/install.
PS C:\Windows\system32>

Tried the following commands, but none of them solved the problem:
dism.exe /online /enable-feature /featurename:Microsoft-Windows-Subsystem-Linux /all /norestart
dism.exe /online /enable-feature /featurename:VirtualMachinePlatform /all /norestart
wsl --update
wsl --set-default-version 2
wsl --install -d Ubuntu

Will not work

dorssel, 2024-07-16:
> PS C:\Windows\system32> usbipd bind --busid 3-4 PS C:\Windows\system32> usbipd attach --wsl --busid 3-4 usbipd: error: Windows Subsystem for Linux version 2 is not available. See https://learn.microsoft.com/windows/wsl/install. PS C:\Windows\system32>

That's a completely different error than reported by the OP. Not related at all. Please file a new issue if you want to resolve it.

WoojungHuh, 2024-08-06:
Create a new issue #1008

dorssel, 2025-02-02:
Stale.

closed 2025-02-02 by dorssel