dorssel/usbipd-win #976: Trouble connecting sd card reader to work with gparted

MithrilMan, 2024-06-21:
Hello.
I need to create partitions for an sd card on linux and I'm on windows.
I've WSL2 with Ubuntu 24.04 LTS
I've installed latest version of usbipd on windows (10), I've bound and attached the reader, this is the result of `usbipd list`

[image]

if I run `lsusb` on ubuntu, it shows the device
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 05e3:0749 Genesys Logic, Inc. SD Card Reader and Writer
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

lsblk shows this
```
sudo lsblk
NAME  MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0   7:0    0 469.3M  1 loop /mnt/wsl/docker-desktop/cli-tools
loop1   7:1    0   152M  1 loop
loop2   7:2    0 419.3M  1 loop
sda     8:0    0 388.6M  1 disk
sdb     8:16   0     6G  0 disk [SWAP]
sdc     8:32   0     1T  0 disk /mnt/wsl/docker-desktop/docker-desktop-user-distro
sdd     8:48   0   256G  0 disk /mnt/wsl/docker-desktop-data/isocache
sde     8:64   0   256G  0 disk /mnt/wslg/distro
```

(I'm not a linux expert but I think something is missing, I was expecting to find my card drive)
if I start `gparted`, it doesn't show my sd drive.

Is this supported? What am I missing?

```
wsl --version
WSL version: 2.2.4.0
Kernel version: 5.15.153.1-2
WSLg version: 1.0.61
MSRDC version: 1.2.5326
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26091.1-240325-1447.ge-release
Windows version: 10.0.19045.4529
```

here WSL version

Thanks

dorssel, 2024-06-22:
I don't think SD drivers are in the default WSL kernel. You probably need to compile your own, with the correct driver for your reader enabled. See https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers

platima, 2024-11-27:
Just cut a kernel for this, as I had the same problem: https://github.com/platima/WSL

closed 2025-02-02 by dorssel