dorssel/usbipd-win #977: usbipd: error: Device is not shared; run 'usbipd bind --busid 5-1' as administrator first

mathmorier, 2024-06-24:
Hello,
I have a problem  I tried to connect my printer to WSL but it's say is not shared any one know I can made my device shared ?

My logs :
```
PS C:\Users\mathm> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
5-1    04a9:32db  Prise en charge d’impression USB, Canon CP1300 Null Driver    Not shared
5-3    04a9:3270  Canon EOS 100D                                                Attached
5-4    1a86:e5e3  Périphérique d’entrée USB                                     Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\mathm> usbipd attach -a --wsl --busid=5-1
usbipd: error: Device is not shared; run 'usbipd bind --busid 5-1' as administrator first.
PS C:\Users\mathm>
```

dorssel, 2024-06-24:
Just after it is detached, what is the output of `usbipd list`? My guess is that you have one of those devices that after initialization resets and becomes a different device (different VID:PID).

mathmorier, 2024-06-25:
I'm will check tonight, but I think not because on my two last message the device still with the same VID:PID 04a9:32db.
Do you have an idea hwo I can fix this problem ?

dorssel, 2024-06-25:
Follow the instruction on https://github.com/dorssel/usbipd-win/wiki/Troubleshooting to get a console log and a USB capture.

mathmorier, 2024-06-25:
Okey that my log when I send this command :
```
PS C:\Users\mathm> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
4-1    04a9:32db  Prise en charge d’impression USB, Canon CP1300 Null Driver    Shared
4-3    04a9:3270  Canon EOS 100D                                                Shared
4-4    1a86:e5e3  Périphérique d’entrée USB                                     Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\mathm> usbipd attach -a --wsl --busid=4-1
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.30.240.1 to reach the host.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
WSL Attached
WSL Detached
WSL Attached
WSL Detached
WSL Attached
WSL Detached
WSL Attached
WSL Detached
WSL Attached
WSL Detached
PS C:\Users\mathm> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
4-1    04a9:32db  Prise en charge d’impression USB, Canon CP1300 Null Driver    Shared
4-3    04a9:3270  Canon EOS 100D                                                Shared
4-4    1a86:e5e3  Périphérique d’entrée USB                                     Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\mathm>
```
My Log :
```
PS C:\Windows\system32> usbipd server Logging:LogLevel:Default=Trace "usbipd:PcapNg:Path=C:\FULL\PATH\TO\FILE.pcapng"
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = 'C:\FULL\PATH\TO\FILE.pcapng'
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:SnapLength = unlimited
fail: Usbipd.PcapNg[4]
      An internal error occurred: Unable to start capture.
      System.IO.DirectoryNotFoundException: Could not find a part of the path 'C:\FULL\PATH\TO\FILE.pcapng'.
         at Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(String, FileMode, FileAccess, FileShare, FileOptions) + 0x151
         at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String, FileMode, FileAccess, FileShare, FileOptions, Int64, Nullable`1) + 0x44
         at System.IO.Strategies.OSFileStreamStrategy..ctor(String, FileMode, FileAccess, FileShare, FileOptions, Int64, Nullable`1) + 0x50
         at System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(String, FileMode, FileAccess, FileShare, FileOptions, Int64, Nullable`1) + 0x6c
         at System.IO.FileStream..ctor(String, FileMode, FileAccess, FileShare, Int32, FileOptions, Int64) + 0x8a
         at Usbipd.PcapNg..ctor(IConfiguration config, ILogger`1 logger) + 0x308
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      4.2.0+54.Branch.master.Sha.5b36605535dd52bc3d8a8e4bbaee7f4d480c86ad
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
[... 164 lines cut ...]
      Client 172.30.255.2 released device at 4-1 (USB\VID_04A9&PID_32DB\B619092710478344).
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.InvalidOperationException: This access control list is not in canonical form and therefore cannot be modified.
         at System.Security.AccessControl.CommonAcl.ThrowIfNotCanonical() + 0x37
         at System.Security.AccessControl.CommonAcl.AddQualifiedAce(SecurityIdentifier, AceQualifier, Int32, AceFlags, ObjectAceFlags, Guid, Guid) + 0x3d
         at System.Security.AccessControl.DiscretionaryAcl.AddAccess(AccessControlType, SecurityIdentifier, Int32, InheritanceFlags, PropagationFlags) + 0x9d
         at System.Security.AccessControl.CommonObjectSecurity.ModifyAccess(AccessControlModification, AccessRule, Boolean&) + 0x28b
         at System.Security.AccessControl.CommonObjectSecurity.AddAccessRule(AccessRule) + 0x49
         at Usbipd.RegistryUtils.SetDeviceAsAttached(Guid, BusId, IPAddress, String) + 0xd6
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0x131f
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0xfe5
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0x1821
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x300
```

Any idea 😃

dorssel, 2024-06-25:
> System.IO.DirectoryNotFoundException: Could not find a part of the path 'C:\FULL\PATH\TO\FILE.pcapng'.

Seriously? You are supposed to provide a storage location and upload the file here...

dorssel, 2024-06-25:
> System.InvalidOperationException: This access control list is not in canonical form and therefore cannot be modified.

You have some sort of corruption in your registry. Maybe uninstalling/reinstalling `usbipd-win` will help (it will recreate the registry entry). Or else you will have to do some repairs. Look through the older issues, I've seen someone with this problem before, and they were able fix it.

dorssel, 2024-06-26:
See https://github.com/dorssel/usbipd-win/issues/218