dorssel/usbipd-win #980: Can find bus device in wsl but when i try running camera code for Intel realsense d435 it says no device connected

YeachanOkh, 2024-07-01:
[image]

I got the bus device to show in wsl but when i try running the camera it seems to not turn on and it give me the error saying that no device connected

[image]

YeachanOkh, 2024-07-01:
it is saying the device is attached so im not sure why the code isnt detecting it 
[image]

dorssel, 2024-07-02:
The default WSL kernel probably does not include a driver for your device. You need to compile your own kernel with the driver included, see https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers