dorssel/usbipd-win #982: How to find my device with generic names

simeoncapy, 2024-07-02:
When I use the `usbipd list` command I cannot see my device, to be exact its name. Indeed, I have 4 generic "USB Input Device", so I don't know how to find it. But my device is recognised with its name in the devices list of Windows settings.
[image]

Here is the result of the command:
```
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-4    1c4f:0027  USB Input Device                                              Not shared
4-1    17ef:609d  USB Input Device                                              Not shared
4-3    06cb:00f9  Synaptics UWP WBDI                                            Not shared
4-4    174f:1812  Integrated Camera, Integrated IR Camera, Camera DFU Device    Not shared
4-7    2988:0304  USB Input Device                                              Not shared
4-8    2ce3:9563  Alcorlink USB Smart Card Reader                               Not shared
4-10   8087:0033  Intel(R) Wireless Bluetooth(R)                                Not shared
6-4    0bda:1101  USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE

usbipd: warning: USB filter 'USBPcap' is known to be incompatible with this software; 'bind --force' will be required.
```
So is there a way to discover which generic device is the one I am looking for?

dorssel, 2024-07-02:
The device descriptions are those reported by the Windows USB subsystem; they are the same (or should be) as in Windows Device Manager. You can display the Linux device names (which may or may not provide more details) using `usbipd list --usbids`.

simeoncapy, 2024-07-03:
Thanks for the answer. I checked on the Device Manager but I couldn't see my device. So I don't understand why the new settings can, and not the Device Manager. Another manifestation of Window's bicephalism...

But thanks for your other command. It couldn't detect my device but reduced the Unknown device to 2. So by unplugging my device, I could discriminate it.

```
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    2988:0304  Unknown device                                                Not shared
2-4    1c4f:0027  SiGma Micro, Unknown device                                   Not shared
4-1    17ef:609d  Lenovo, Unknown device                                        Not shared
4-3    06cb:00f9  Synaptics, Inc., Unknown device                               Not shared
4-4    174f:1812  Syntek, Unknown device                                        Not shared
4-8    2ce3:9563  Unknown device                                                Not shared
4-10   8087:0033  Intel Corp., AX211 Bluetooth                                  Not shared
6-4    0bda:1101  Realtek Semiconductor Corp., Unknown device                   Not shared
```

And on WSL it appears with its name:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2988:0304 3D Systems 3D Systems Haptic Device
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dorssel, 2024-07-03:
Are you sure the default WSL kernel has a driver for your device? You may need to compile your own, see https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers

closed 2025-01-27 by dorssel