dorssel/usbipd-win #983: Connecting a Intel RealSense Depth Camera D415

Mokaz, 2024-07-03:
I'm trying to connect a RealSense D415 Depth Camera, but it does not attach correctly. I have tried using bind with `--force`, but it still does not show up in WSL when using `lsusb`
[image]

Running `usbipd attach --wsl --busid 2-17`:
[image]

Firewall rule inbound:
[image]

Firewall rule outbound:
[image]

I'm running Ubuntu 22.04.4 LTS

dorssel, 2024-07-03:
What is the version of usbipd-win? And of your Windows host?

Mokaz, 2024-07-04:
usbipd-win 4.2.0 and Windows 11 Pro 23H2 22631.3737

dorssel, 2024-07-04:
Are you using 'networkingMode=mirrored' or the default bridged mode?

Mokaz, 2024-07-04:
Default bridged mode. I haven't changed any settings.

dorssel, 2024-07-04:
Are you using any third party security software? Something seems to be blocking the network connection from WSL back to the host.

Mokaz, 2024-07-05:
Well yes, that could very well be the problem. I'm using a work pc, and there are IT security programs installed. I'll contact IT and ask about it. If unblocking the ports is not an option, do you have any other suggestions for connecting the camera to WSL?

dorssel, 2024-07-06:
You can use an `ssh` reverse tunnel. Look through older (possible closed) issues about this subject.

Mokaz, 2024-07-13:
Turns out company IT policy was blocking usbipd. After removing restrictions the device attaches fine, and shows up WSL using lsusb. However, it is not detected by Intel Realsense viewer.

closed 2024-07-13 by Mokaz