dorssel/usbipd-win #988: After the iPhone bind and attach are successful, the lsusb cannot be found in the WSL system

lucky2046, 2024-07-06:
Start PowerShell in Administrator mode

`
PS C:\Windows\system32> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-2    046d:c05a  USB 输入设备                                                  Not shared
3-2    05ac:12a8  Apple Mobile Device USB Composite Device                      Not shared
4-1    8087:0029  英特尔(R) 无线 Bluetooth(R)                                   Not shared
4-2    1bcf:28c4  Integrated Camera                                             Not shared
4-3    048d:8910  USB 输入设备                                                  Not shared
4-4    258a:002a  USB 输入设备                                                  Not shared

Persisted:
GUID                                  DEVICE

usbipd: warning: Unknown USB filter 'nxusbf' may be incompatible with this software; 'bind --force' may be required.
PS C:\Windows\system32> usbipd bind --force --busid 3-2
PS C:\Windows\system32> usbipd attach --wsl --busid=3-2
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.30.96.1 to reach the host.
`

Can't see iPhone device in wsl

`
root@DESKTOP-4VON23E:/mnt/c/Windows/system32# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

Please help me find out what the problem is？

dorssel, 2024-07-06:
Post the debug logs as explained here: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

Probably the device is detached immediately after attach due to a missing driver.

lucky2046, 2024-07-06:
usbipd list show  iPhone  shared

lucky2046, 2024-07-06:
> Post the debug logs as explained here: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting
> 
> Probably the device is detached immediately after attach due to a missing driver.
> 
> EDIT: also post the output of the command `usbipd list` right before and right after attachment.

PS C:\Windows\system32> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-2    046d:c05a  USB 输入设备                                                  Not shared
3-2    05ac:12a8  Apple Mobile Device USB Composite Device                      Shared (forced)
4-1    8087:0029  英特尔(R) 无线 Bluetooth(R)                                   Not shared
4-2    1bcf:28c4  Integrated Camera                                             Not shared
4-3    048d:8910  USB 输入设备                                                  Not shared
4-4    258a:002a  USB 输入设备                                                  Not shared

Persisted:
GUID                                  DEVICE

usbipd: warning: Unknown USB filter 'nxusbf' may be incompatible with this software; 'bind --force' may be required.

dorssel, 2024-07-06:
Can you post the PcapNg?

lucky2046, 2024-07-06:
> Can you post the PcapNg?

not yet，I recreated the wsl ubuntu system and was able to successfully map the iPhone to the wsl subsystem

dorssel, 2024-07-06:
So, the problem is solved (for now)?

lucky2046, 2024-07-06:
> So, the problem is solved (for now)?

Yes, Thank you.

closed 2024-07-06 by lucky2046