dorssel/usbipd-win #990: usbipd not being recognised as command or executable

Deus-Ex-Machina2, 2024-07-11:
I am knew to embedded systems and I am unsure why I cannot use this software. I keep getting the following error:

usbipd list

usbipd : The term 'usbipd' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.

At line:1 char:1
+ usbipd list
+ ~~~~~~
    + CategoryInfo          : ObjectNotFound: (usbipd:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
    
My path environment variable is C:\Program Files\usbipd-win\

Please could someone assist, as I do not know what is wrong.

dorssel, 2024-07-12:
what is the output of `dir "C:\Program Files\usbipd-win"` ?

Deus-Ex-Machina2, 2024-07-12:
> what is the output of `dir "C:\Program Files\usbipd-win"` ?

Hi dorssel, I end-up using the .msi file again to do repairs. So the problem is solved now. Thanks for getting back to me though.

closed 2024-07-12 by Deus-Ex-Machina2