dorssel/usbipd-win #992: BSOD in 4.2.0

Nygrki, 2024-07-17:
Hi,

im facing similiar issues than in the issue before https://github.com/dorssel/usbipd-win/issues/717.
In my scenario the BSOD happens every time I try to attach the usb device to wsl2.

@dorssel any ideas what might cause this issue?
I tried versions 4.0.0, 4,1.0 and the latest 4.2.0, because I saw there were driver updates in every update but without any luck. Is this a driver issue, or is there something wrong with my system?  

Below is the crash analysis from WinDBG.
`PNP_DETECTED_FATAL_ERROR (ca)
PnP encountered a severe error, either as a result of a problem in a driver or
a problem in PnP itself.  The first argument describes the nature of the
problem, the second argument is the address of the PDO.  The other arguments
vary depending on argument 1.
Arguments:
Arg1: 0000000000000002, Invalid PDO
	An API which requires a PDO has been called with either an FDO,
	a PDO which hasn't been initialized yet (returned to PnP in a
	QueryDeviceRelation/BusRelations), or some random piece of
	memory.
Arg2: ffffe7836003d9e0, Purported PDO.
Arg3: 0000000000000000, Driver object.
Arg4: 0000000000000000

Debugging Details:
------------------

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 4406

    Key  : Analysis.Elapsed.mSec
    Value: 18488

    Key  : Analysis.IO.Other.Mb
    Value: 0

    Key  : Analysis.IO.Read.Mb
    Value: 0

    Key  : Analysis.IO.Write.Mb
    Value: 1

    Key  : Analysis.Init.CPU.mSec
    Value: 608

    Key  : Analysis.Init.Elapsed.mSec
    Value: 23829

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 99

    Key  : Bugcheck.Code.LegacyAPI
    Value: 0xca

    Key  : Bugcheck.Code.TargetModel
    Value: 0xca

    Key  : Failure.Bucket
    Value: 0xCA_2_VRF_VBoxUSBMon!unknown_function

    Key  : Failure.Hash
    Value: {7e015e92-49a9-80e2-3e1d-d1242f1ceb8c}

    Key  : Hypervisor.Enlightenments.Value
    Value: 68673420

    Key  : Hypervisor.Enlightenments.ValueHex
    Value: 417df8c

    Key  : Hypervisor.Flags.AnyHypervisorPresent
    Value: 1

    Key  : Hypervisor.Flags.ApicEnlightened
    Value: 0

    Key  : Hypervisor.Flags.ApicVirtualizationAvailable
    Value: 1

    Key  : Hypervisor.Flags.AsyncMemoryHint
    Value: 0

    Key  : Hypervisor.Flags.CoreSchedulerRequested
    Value: 0

    Key  : Hypervisor.Flags.CpuManager
    Value: 1

    Key  : Hypervisor.Flags.DeprecateAutoEoi
    Value: 1

    Key  : Hypervisor.Flags.DynamicCpuDisabled
    Value: 1

    Key  : Hypervisor.Flags.Epf
    Value: 0

    Key  : Hypervisor.Flags.ExtendedProcessorMasks
    Value: 1

    Key  : Hypervisor.Flags.HardwareMbecAvailable
    Value: 1

    Key  : Hypervisor.Flags.MaxBankNumber
    Value: 0

    Key  : Hypervisor.Flags.MemoryZeroingControl
    Value: 0

    Key  : Hypervisor.Flags.NoExtendedRangeFlush
    Value: 0

    Key  : Hypervisor.Flags.NoNonArchCoreSharing
    Value: 1

    Key  : Hypervisor.Flags.Phase0InitDone
    Value: 1

    Key  : Hypervisor.Flags.PowerSchedulerQos
    Value: 0

    Key  : Hypervisor.Flags.RootScheduler
    Value: 0

    Key  : Hypervisor.Flags.SynicAvailable
    Value: 1

    Key  : Hypervisor.Flags.UseQpcBias
    Value: 0

    Key  : Hypervisor.Flags.Value
    Value: 21631230

    Key  : Hypervisor.Flags.ValueHex
    Value: 14a10fe

    Key  : Hypervisor.Flags.VpAssistPage
    Value: 1

    Key  : Hypervisor.Flags.VsmAvailable
    Value: 1

    Key  : Hypervisor.RootFlags.AccessStats
    Value: 1

    Key  : Hypervisor.RootFlags.CrashdumpEnlightened
    Value: 1

    Key  : Hypervisor.RootFlags.CreateVirtualProcessor
    Value: 1

    Key  : Hypervisor.RootFlags.DisableHyperthreading
    Value: 0

    Key  : Hypervisor.RootFlags.HostTimelineSync
    Value: 1

    Key  : Hypervisor.RootFlags.HypervisorDebuggingEnabled
    Value: 0

    Key  : Hypervisor.RootFlags.IsHyperV
    Value: 1

    Key  : Hypervisor.RootFlags.LivedumpEnlightened
    Value: 1

    Key  : Hypervisor.RootFlags.MapDeviceInterrupt
    Value: 1

    Key  : Hypervisor.RootFlags.MceEnlightened
    Value: 1

    Key  : Hypervisor.RootFlags.Nested
    Value: 0

    Key  : Hypervisor.RootFlags.StartLogicalProcessor
    Value: 1

    Key  : Hypervisor.RootFlags.Value
    Value: 1015

    Key  : Hypervisor.RootFlags.ValueHex
    Value: 3f7

    Key  : WER.OS.Branch
    Value: vb_release

    Key  : WER.OS.Version
    Value: 10.0.19041.1

BUGCHECK_CODE:  ca

BUGCHECK_P1: 2

BUGCHECK_P2: ffffe7836003d9e0

BUGCHECK_P3: 0

BUGCHECK_P4: 0

FILE_IN_CAB:  071724-21453-01.dmp

DEVICE_OBJECT: ffffe7836003d9e0

DRIVER_OBJECT: ffffe7835ef87df0

IMAGE_NAME:  VBoxUSBMon.sys

FAULTING_MODULE: fffff8054f800000 UsbHub3

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

CUSTOMER_CRASH_COUNT:  1

PROCESS_NAME:  System

STACK_TEXT:  
ffffe6888ac4e368 fffff8065504b0f9     : 00000000000000ca 0000000000000002 ffffe7836003d9e0 0000000000000000 : nt!KeBugCheckEx
ffffe6888ac4e370 fffff806551e1c23     : ffffe78381414fe0 ffffe7836003d9e0 ffffe783814128e0 fffff805ba494400 : nt!IoGetDevicePropertyData+0x130699
ffffe6888ac4e3c0 fffff805ba4939e8     : ffffe78381414fe0 ffffe7836003d9e0 ffffe783814128e0 ffffe6888ac4e348 : nt!VerifierIoGetDevicePropertyData+0x43
ffffe6888ac4e420 ffffe78381414fe0     : ffffe7836003d9e0 ffffe783814128e0 ffffe6888ac4e348 0000000000000600 : VBoxUSBMon+0x39e8
ffffe6888ac4e428 ffffe7836003d9e0     : ffffe783814128e0 ffffe6888ac4e348 0000000000000600 ffffe783814129f4 : 0xffffe78381414fe0
ffffe6888ac4e430 ffffe783814128e0     : ffffe6888ac4e348 0000000000000600 ffffe783814129f4 ffffe6888ac4e464 : 0xffffe7836003d9e0
ffffe6888ac4e438 ffffe6888ac4e348     : 0000000000000600 ffffe783814129f4 ffffe6888ac4e464 ffffe6888ac4e460 : 0xffffe783814128e0
ffffe6888ac4e440 0000000000000600     : ffffe783814129f4 ffffe6888ac4e464 ffffe6888ac4e460 ffffe6888ac50000 : 0xffffe6888ac4e348
ffffe6888ac4e448 ffffe783814129f4     : ffffe6888ac4e464 ffffe6888ac4e460 ffffe6888ac50000 ffffe6888ac49000 : 0x600
ffffe6888ac4e450 ffffe6888ac4e464     : ffffe6888ac4e460 ffffe6888ac50000 ffffe6888ac49000 0000000000000000 : 0xffffe783814129f4
ffffe6888ac4e458 ffffe6888ac4e460     : ffffe6888ac50000 ffffe6888ac49000 0000000000000000 ffffe6888ac50000 : 0xffffe6888ac4e464
ffffe6888ac4e460 ffffe6888ac50000     : ffffe6888ac49000 0000000000000000 ffffe6888ac50000 0000000000000000 : 0xffffe6888ac4e460
ffffe6888ac4e468 ffffe6888ac49000     : 0000000000000000 ffffe6888ac50000 0000000000000000 ffffe7835880d798 : 0xffffe6888ac50000
ffffe6888ac4e470 0000000000000000     : ffffe6888ac50000 0000000000000000 ffffe7835880d798 ffffd634936b3b57 : 0xffffe6888ac49000

SYMBOL_NAME:  VBoxUSBMon+39e8

MODULE_NAME: VBoxUSBMon

STACK_COMMAND:  .cxr; .ecxr ; kb

BUCKET_ID_FUNC_OFFSET:  39e8

FAILURE_BUCKET_ID:  0xCA_2_VRF_VBoxUSBMon!unknown_function

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {7e015e92-49a9-80e2-3e1d-d1242f1ceb8c}

Followup:     MachineOwner
---------`

dorssel, 2024-07-17:
> IMAGE_NAME:  VBoxUSBMon.sys

It is indeed a duplicate of #717. This needs to be fixed in VirtualBox, better report it there (if it hasn't already been reported).

closed 2024-07-17 by dorssel

dennisafa-slingshot, 2024-07-20:
Just wanted to note that I am having the exact same crash occur on a Windows Server 2019 machine.