dorssel/usbipd-win #995: `WSL kernel is not USBIP capable` with WSL pre-release 2.3.11

polyzen, 2024-07-17:
I can't seem to find anything matching `vhci` nor `usbip` under `/sys`. Not sure if there's some place else to check or what.

`sudo usbip list --remote=$IP` results in `usbip: error: could not connect to $IP:3240: System error`

Distro is Arch Linux.

dorssel, 2024-07-17:
On your WSL instance, what is the output of `cat /sys/devices/platform/vhci_hcd.0/status`?

polyzen, 2024-07-17:
On July 17, 2024 7:28:12 PM EDT, Frans van Dorsselaer ***@***.***> wrote:
> On your WSL instance, what is the output of `cat /sys/devices/platform/vhci_hcd.0/status`?
> 

That directory does not exist.

--
Best,
Daniel <https://danielcapella.com>

dorssel, 2024-07-17:
What is the output of `zgrep CONFIG_USBIP /proc/config.gz`?

polyzen, 2024-07-17:
<details>
  <summary>show/hide</summary>
  <p>

```
CONFIG_USBIP_CORE=m
CONFIG_USBIP_VHCI_HCD=m
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
CONFIG_USBIP_HOST=m
# CONFIG_USBIP_DEBUG is not set
```
  </p>
</details>

dorssel, 2024-07-17:
Ah, they modularized it! (`=m`); this used to be monolithically built-in (`=y`).
You need to load the module first: `sudo modprobe vhci-hcd`.

polyzen, 2024-07-18:
Works now :D

polyzen, 2024-07-18:
https://wiki.archlinux.org/title/Kernel_module#systemd docs for automatic module loading

Should I leave this open until this is documented or something?

dorssel, 2024-07-18:
No, but I would report this to https://github.com/microsoft/WSL/issues. I don't think it is intended. But if it is, we would like to know that too. It has happened in the past that they inadvertently put some config settings to "module", whereas the WSL kernel (at least until now) is supposed to be entirely monolithic.

dorssel, 2024-07-18:
BTW: it worthwhile noting thay `CONFIG_USBIP_HOST=m` makes absolutely no sense. It used to be `# CONFIG_USBIP_HOST is not set`. WSL can never be a USB host. There is no physical USB, and the Linux USBIP subsystem does not allow forwarding client USB devices.

polyzen, 2024-07-18:
https://github.com/microsoft/WSL/issues/11792

dorssel, 2024-07-18:
From the release notes it appears they *are* going the modularized way... No definitive answer yet, see https://github.com/microsoft/WSL/discussions/11789#discussioncomment-10083621

In the meanwhile, I've attempted supporting it, see #996.

Could you give the development installer a try? It is at https://github.com/dorssel/usbipd-win/actions/runs/9997393400?pr=996

Please test:
- clean WSL boot (`wsl --shutdown` first)
- without loading the module manually: `usbipd attach --wsl`
- `usbipd detach`
- while the module is already loaded: `usbipd attach --wsl`

Please post the console output. Thx.

polyzen, 2024-07-18:
```
~ > usbipd.exe attach --wsl --hardware-id '$hwid'
usbipd: info: Device with hardware-id '$hwid' found at busid '$busid'.
usbipd: info: Using WSL distribution 'Arch' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Loading vhci_hcd module.
usbipd: info: Using IP address $ip to reach the host.
~ > usbipd.exe detach --hardware-id '$hwid'
usbipd: info: Device with hardware-id '$hwid' found at busid '$busid'.
~ > usbipd.exe attach --wsl --hardware-id '$hwid'
usbipd: info: Device with hardware-id '$hwid' found at busid '$busid'.
usbipd: info: Using WSL distribution 'Arch' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address $ip to reach the host.

dorssel, 2024-07-18:
@polyzen 
Awesome, thx. I'm merging this, which will (once again) close this ticket. Thanks for the input!

PR #996 (2024-07-18): Support WSL kernel modules
Fixes #995
closed 2024-07-18 by dorssel