dorssel/usbipd-win #997: WSL usbip: error: tcp connect - Trying to passthrough USB to use in a docker devcontainer (for STM32 debugging)

yusufso, 2024-07-19:
Here are the commands I run and the corresponding output.

`
> usbipd bind --force --busid=2-2
> usbipd attach --wsl --busid=2-2
usbipd: info: Using WSL distribution 'docker-desktop' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.18.96.1 to reach the host.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '2-2'.
`

The devcontainer & docker engine were already running when I ran the snippet above. I used the .msi to install USBIPD. Running on Windows 10 Pro Version 10.0.19045 Build 19045. The device I am trying to passthrough is an STLINK-V3 debugger used for STM32 microcontrollers.

Is there an option for USBIPD to print out debug info? The above snippet doesn't feel like enough to diagnose the issue. Please tell me what other information you may need.

dorssel, 2024-07-19:
That is weird indeed, as there is a connectivity check even before `usbip` is called client side. Debugging (https://github.com/dorssel/usbipd-win/wiki/Troubleshooting) won't help here, as the error happens on the Linux side.

However, the connectivity check is very forgiving. My bet is that it somehow failed, but gives a false positive. And then when `usbip` is called, the real TCP connection to port 3240 fails. Are you using third party security software (firewall), or are there company domain policies in effect?

The connectivity check uses `bash`, which I believe is not available in `docker-desktop`. That may be the reason the connectivity test is not even executed (which is treated as a no-fail, as it is just optional). Somehow, incoming connections on port 3240 are blocked.

dorssel, 2024-07-20:
Can you try the installer from #998? It is at https://github.com/dorssel/usbipd-win/actions/runs/10015693449?pr=998.

yusufso, 2024-07-24:
I tried it and it worked, took a little while for the command to finish though.

I also tried v4.2.0 again but with windows 11, that time it worked.

By worked I mean you can see the usb device after "lsusb". I still have to determine if the usb device works as expected.

I will close the issue now.

closed 2024-07-24 by yusufso