dorssel/usbipd-win #1002: Windows PowerShell Error: "usbipd: warning: The service is currently not running; a reboot should fix that."

jlapp12, 2024-07-25:
Hello,

I am trying to use this to interact on WSL2 using Ubuntu version 5.15.153.1. I am installed to the nearest version, I have an x64 system, and I have previously downloaded and tried repairing using the installer.

When I attempt to use the usbipd --list command in the Windows PowerShell as an administrator, I keep receiving the error:  "usbipd: warning: The service is currently not running; a reboot should fix that." I have binded a device, but when I try to attach, again, the same error occurs.

I have restarted/shutdown multiple times to no success. Is there something else I can try to fix this? I have tried looking at similar issues, to no avail.

Thank you.

dorssel, 2024-07-25:
Try a repair of the software, or else an uninstall/re-install.

jlapp12, 2024-07-25:
Thank you for the quick response. I have tried both a repair and an uninstall and re-install. This has not fixed this issue. Is there a way I can run the service manually?

dorssel, 2024-07-26:
yes, see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

jlapp12, 2024-07-26:
OK -- I have run the start command for the server (sc.exe start usbipd), and I have been given the error of: "[SC] StartService FAILED 1053: The service did not respond to the start or control request in a timely fashion."

I have read here: https://answers.microsoft.com/en-us/windows/forum/all/error-1053-custom-service-does-not-start/810fcf93-0cca-43cd-9e71-9050e7ee80ed that I could use the NSSM.exe and install the service name? If I am making any sense, what would the service name be for this? Is this a possible fix?

My apologizes for my ignorance.

jlapp12, 2024-07-26:
After some further digging, I tried a video solution for error 1053, and it worked. Not sure how what he does works, but he goes into Registry Editor, HKEY_LOCAL_MACHINE --> SYSTEM --> CurrentControlSet --> Control and creates a new DWORD (32-bit) called ServicesPipeTimeout with a hexadecimal value data of 18000. After a restart the issue resolved. https://www.youtube.com/watch?v=QPKGZ2ja8-I

closed 2024-07-26 by jlapp12