dorssel/usbipd-win #1003: Unable to attach USB to WSL-Ubuntu-22.04

SSaiPrasad7, 2024-07-26:
Hello @dorssel 

I'm attempting to connect a USB device to my WSL Ubuntu setup, but it's not working. Could you assist me with this issue? I've attached screenshots to aid in troubleshooting.

Thank you in advance 

- Windows:
[image]
- WSL-Ubuntu-22.04:
[image]

timsu92, 2024-07-27:
I guess Microsoft may have changed something in the new kernel. I was running on the official kernel version 5.15.133 and I'm running on a customized one with UVC camera support compiled by myself.
Maybe you can head over [Microsoft's kernel page](https://github.com/microsoft/WSL2-Linux-Kernel) and try to compile to see if the problem comes from the new kernel.

dorssel, 2024-07-27:
> I guess Microsoft may have changed something in the new kernel.

They did, duplicate of: https://github.com/dorssel/usbipd-win/issues/995 (with solution)

closed 2024-07-27 by dorssel