dorssel/usbipd-win #1005: adb pushing large file hangs

natsu1375, 2024-08-02:
usbipd version: [4.2.0+54.Branch.master.Sha.5b36605535dd52bc3d8a8e4bbaee7f4d480c86ad]
windows build: [windows 11 (Version 10.0.22631 Build 22631)]
wsl kernel version: [5.15.153.1-2]

It's ok with small file
[image]

But just hanged when pushing an apk file
[image]

Each pushing resulted in a different percentage.

dorssel, 2024-08-03:
See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. A capture could help.

herenfor, 2026-10-22:
Another data point for adb bulk stalls on WSL kernel 6.18, which looks like a regression from 5.15 in my setup.

Same usbipd-win and same adb on both kernels:
- usbipd-win 5.3.0-54+Branch.master.Sha.aa3db8b82c4cb5071fd31bc54211606c70886912
- adb from Android platform-tools 37.0.1 (inside WSL)
- Device: Samsung phone (SM-S9370), USB debugging, attached with `usbipd attach --wsl`
- Windows: <fill in build>

Before (WSL kernel 5.15.167.4-microsoft-standard-WSL2): `adb install` of ~20 MB APKs worked repeatedly over several weeks (one install hung once and succeeded on retry).

After (WSL 3.0.3.0, kernel 6.18.54.1-microsoft-standard-WSL2; nothing else changed on the USB path): host→device transfers stall deterministically by size:
- adb shell commands, port forwarding / WebView DevTools: fine
- device→host is fine (e.g. `adb exec-out screencap -p`, ~260 KB)
- `adb push` of 100 KB: fine; 500 KB, 1 MB, 2 MB, 8 MB: hangs every time (no progress, device stays "Attached", `adb devices` still lists it, small commands keep working afterwards)
- Same result with ADB_LIBUSB=0 and ADB_LIBUSB=1

Workaround: split the file into 100 KB chunks, push each, `cat` them together on the device (sha256 matches), then `pm install`. 215 chunks pushed in ~8 s without a single failure.

I haven't captured pcapng/usbmon yet; happy to do so if that helps.