dorssel/usbipd-win #1008: "usbipd: error: There are no WSL distributions installed." when "usbipd attach --wsl --busid xx-yy"

WoojungHuh, 2024-08-06:
Versios info by "usbipd --version"
4.3.0-70+Branch.master.Sha.701757afaea0ed252dd541f290decafe807970a5.701757afaea0ed252dd541f290decafe807970a5

Device is shared with "--force" as instructed because USBPcap imcompatible.
```
> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-1    045e:083e  Modern USB-C Speaker, USB Input Device                        Not shared
2-6    0c45:6720  Integrated Webcam                                             Not shared
2-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-1    291a:3361  Anker PowerConf C300                                          Not shared
4-4    0bda:8153  Realtek USB GbE Family Controller                             Not shared
8-5    413c:b06e  USB Input Device                                              Not shared
10-1   04d8:00df  USB Serial Device (COM3), USB Input Device                    Shared (forced)
10-5   413c:b06f  USB Input Device                                              Not shared
12-1   0a12:1004  AfterShokz Loop 100, USB Input Device                         Not shared
```

WSL version info 
```
> wsl --version
WSL version: 2.2.4.0
Kernel version: 5.15.153.1-2
WSLg version: 1.0.61
MSRDC version: 1.2.5326
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26091.1-240325-1447.ge-release
Windows version: 10.0.22631.3880
```
Got error as below.
```
> usbipd attach --wsl -b 10-1
usbipd: error: There are no WSL distributions installed. Learn how to install one at https://learn.microsoft.com/windows/wsl/basic-commands#install.
```

dorssel, 2024-08-07:
From the very same terminal where the `usbipd attach --wsl` command is failing, what is the output of `wsl --list --verbose`?

WoojungHuh, 2024-08-07:
Hmm. Weird. I'm using WSL, but no distribution with that command.

```
> wsl --list --verbose
Windows Subsystem for Linux has no installed distributions.

Use 'wsl.exe --list --online' to list available distributions
and 'wsl.exe --install <Distro>' to install.

Distributions can also be installed by visiting the Microsoft Store:
https://aka.ms/wslstore
Error code: Wsl/WSL_E_DEFAULT_DISTRO_NOT_FOUND
```

```
Welcome to Ubuntu 22.04.4 LTS (GNU/Linux 5.15.153.1-microsoft-standard-WSL2 x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/pro

 * Strictly confined Kubernetes makes edge and IoT secure. Learn how MicroK8s
   just raised the bar for easy, resilient and secure K8s cluster deployment.

   https://ubuntu.com/engage/secure-kubernetes-at-the-edge

This message is shown once a day. To disable it please create the
/home/wjhuh/.hushlogin file.

Welcome to #######. Load: 1.33 0.29 0.10 4/302 668
```

dorssel, 2024-08-07:
From the very same terminal where the `usbipd attach --wsl` command is failing, what is the output of `wsl`?

WoojungHuh, 2024-08-07:
device 10-1 is shared
`10-1   04d8:00df  USB Serial Device (COM3), USB Input Device                    Shared (forced)`

Output of the command per your request.
```
> usbipd attach --wsl --busid 10-1
usbipd: error: There are no WSL distributions installed. Learn how to install one at https://learn.microsoft.com/windows/wsl/basic-commands#install.
```

dorssel, 2024-08-07:
No, I mean the literal command `wsl` from the terminal where `usbipd attach --wsl --busid 10-1` is failing.

WoojungHuh, 2024-08-07:
Installed WSL distribution from Microsoft Store than "wsl --install" in my case.
Would it cause this issue?

WoojungHuh, 2024-08-07:
here is it.
```
> wsl
Windows Subsystem for Linux has no installed distributions.

Use 'wsl.exe --list --online' to list available distributions
and 'wsl.exe --install <Distro>' to install.

Distributions can also be installed by visiting the Microsoft Store:
https://aka.ms/wslstore
Error code: Wsl/Service/CreateInstance/GetDefaultDistro/WSL_E_DEFAULT_DISTRO_NOT_FOUND
```

dorssel, 2024-08-07:
It appears you running are the terminal as a different user then the one that installed the Linux distribution.
What is the output of `$env:USERNAME` (PowerShell) or `echo %USERNAME%` (Command Prompt)? Does that match your expected username?

WoojungHuh, 2024-08-07:
Got it. 
Run Powershell in administration mode. Will try in non-admin mode.

WoojungHuh, 2024-08-07:
My bad not reading full lines of instruction.
Thought attach/detach can be run in admin, but attach/detach MUST be executed in non-admin mode.

WoojungHuh, 2024-08-07:
Thanks @dorssel. Issue is resolved.
- run bind/unbind in admin-mode
- run attach/detail in NON admin-mode

The device can be seen and work in WSL without issue.

closed 2024-08-07 by WoojungHuh