dorssel/usbipd-win #1010: usbipd error loading vhci_hcd failed

LZY-SPCA, 2024-08-08:
usbipd error loading vhci_hcd failed while attach usb device to wsl

LZY-SPCA, 2024-08-08:
usbipd-win version
[image]
windows 11 23H2

dorssel, 2024-08-08:
Within WSL, what is the output of `zgrep CONFIG_USBIP /proc/config.gz`?

michal5520pl, 2024-08-09:
@dorssel
`> zgrep CONFIG_USBIP /proc/config.gz
CONFIG_USBIP_CORE=m
CONFIG_USBIP_VHCI_HCD=m
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
CONFIG_USBIP_HOST=m
# CONFIG_USBIP_DEBUG is not set`

I have the same problem with WSL2 pre-release
`Wersja podsystemu WSL: 2.3.14.0
Wersja jądra: 6.6.36.3-1
Wersja usługi WSLg: 1.0.64
Wersja MSRDC: 1.2.5326
Wersja Direct3D: 1.611.1-81528511
Wersja DXCore: 10.0.26100.1-240331-1435.ge-release
Wersja systemu Windows: 10.0.22631.3880`

Apologies for Polish, but I hope it's readable

PS. I have the same version as @LZY-SPCA

dorssel, 2024-08-09:
What happens if you load the module by hand: `sudo modprobe vhci_hcd`?

michal5520pl, 2024-08-09:
No module vhci_vhd found, unfortunately. Did Microsoft some fail regarding that module? I will try to compare versions, maybe

dorssel, 2024-08-10:
Normal (non-pre-release) versions of WSL don't use modules, they have all drivers compiled in a monolithic kernel.

The pre-release version of WSL is switching to modules (`CONFIG_USBIP_CORE=m`), so you will have to load `vhci_hcd`. `usbipd-win` will try to do that automatically. If you don't have that module, there is nothing I can do about it.

LZY-SPCA, 2024-08-10:
I ran modprobe vhci_hcd successfully and solved this problem.

michal5520pl, 2024-08-10:
I've tried again and modprobe vhci_hcd indeed worked. Probably I have mismatched one letter due to tiredness. I have resolved the problem with putting .conf file in /etc/modules-load.d with vhci_hcd inside

closed 2024-08-10 by LZY-SPCA

kingsoftech, 2024-08-26:
hello everyone am currently facing the same issue with the following output 
CONFIG_USBIP_CORE=m
CONFIG_USBIP_VHCI_HCD=m
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
# CONFIG_USBIP_HOST is not set
# CONFIG_USBIP_DEBUG is not set

Morrigan-Ship, 2024-10-03:
same

sudo modprobe vhci_hcd:

CONFIG_USBIP_CORE=m
CONFIG_USBIP_VHCI_HCD=m
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1

Morrigan-Ship, 2024-10-03:
fixed! solved!

run command as Administrator :/

zerobinhi, 2025-03-29:
# Compile the WSI Linux kernel
```bash
make menuconfig KCONFIG_CONFIG=Microsoft/config-wsl
```

（1）进入 Device Drivers -> USB Support

（2）找到 Support for Host-side USB，键入 “y”

（3）找到 USB/IP Support -> VCHI hcd.

[image]

（4）找到 USB Mass Storage support，将其子条目全部选中

[image]
用这个新编译的内核即可

[image]
[参考链接](https://www.cnblogs.com/guan88/articles/18619901)

MightyM17, 2025-06-19:
> ~$ uname -r
> 6.6.87.1-microsoft-standard-WSL2
> 
> the output from powershell in win11 is
> 
> $ wsl --version
> WSL 版本: 2.5.8.0
> 内核版本: 6.6.87.1-1
> WSLg 版本: 1.0.66
> MSRDC 版本: 1.2.6074
> Direct3D 版本: 1.611.1-81528511
> DXCore 版本: 10.0.26100.1-240331-1435.ge-release
> Windows: 10.0.26100.4061
> 
> You can fix above problems by updating your wsl version.

I cannot update my wsl kernel version
```
wsl --version
WSL version: 2.5.9.0
Kernel version: 6.6.87.2-1
WSLg version: 1.0.66
MSRDC version: 1.2.6074
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
Windows version: 10.0.26100.4351
```

```
mighty@mighty:$ sudo modprobe vhci-hcd
[sudo] password for mighty:
modprobe: FATAL: Module vhci-hcd not found in directory /lib/modules/6.6.36.6-microsoft-standard-WSL2+
mighty@mighty:$ zgrep CONFIG_USBIP /proc/config.gz
CONFIG_USBIP_CORE=m
CONFIG_USBIP_VHCI_HCD=m
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
CONFIG_USBIP_HOST=m
# CONFIG_USBIP_DEBUG is not set
mighty@mighty:$ uname -r
6.6.36.6-microsoft-standard-WSL2+
```

Will try building the WSL kernel now

zerobinhi, 2025-06-19:
> I cannot update my wsl kernel version
> Will try building the WSL kernel now
@MightyM17 
To be frank, the recognition of removable storage media is already supported in WSL2 2.5.9.
[WSL2](https://github.com/microsoft/WSL/releases/tag/2.5.9)

ivbaklan2012, 2025-07-24:
> I cannot update my wsl kernel version
>
> ```
> wsl --version
> WSL version: 2.5.9.0
> Kernel version: 6.6.87.2-1
> WSLg version: 1.0.66
> MSRDC version: 1.2.6074
> Direct3D version: 1.611.1-81528511
> DXCore [...]

try `sudo modprobe vhci_hcd` and not vhci-hcd