dorssel/usbipd-win #1011: Device not properly mapped into default docker-desktop distro

Clockwork-Muse, 2024-08-09:
I'm using devcontainers with docker desktop, and mounting in a barcode reader (a Datalogic GFS4520, as a USB-COM serial device).
I can mount the device, and it shows up with `ls` (and presumably `lsusb`), however it mounts as:
```bash
crw-------    1 root     root      166,   0 Aug  9 20:32 /dev/ttyACM0
```
... and not the proper
```bash
crw-rw----    1 root     dialout   166,   0 Aug  9 20:39 /dev/ttyACM0
```
... which then causes various problems inside my devcontainer.
I **can** work around this by attaching to another distro (in this case a manual Ubuntu installation).  However this is obviously not ideal.

When mounting the device the (anticipated) no-bash error is printed, but resolving this by installing bash doesn't solve the permissions/ownership error.
```bash
usbipd: info: Device with hardware-id '05f9:4204' found at busid '2-2'.
usbipd: info: Selecting a specific distribution is no longer required. Please file an issue if you believe that the default selection mechanism is not working for you.
usbipd: info: Using WSL distribution 'docker-desktop' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.27.224.1 to reach the host.
usbipd: info: Firewall check not possible with this distribution (no bash, or wrong version of bash).
```

dorssel, 2024-08-09:
This is not a `usbipd-win` problem. Probably something to do with systemd or udev. As soon as `lsusb` lists the device, `usbipd-win` is done.

Clockwork-Muse, 2024-08-10:
.... ah, could be because `systemd` isn't enabled (by default) in the docker desktop distro, then, I suppose.
Not sure about whatever udev rules....

closed 2024-08-10 by Clockwork-Muse

reynoldsbd, 2024-11-18:
@dorssel I encountered a similar issue with another (non-Docker) WSL distro. It seems device nodes are not created and/or udev rules are not applied unless systemd is running in the WSL2 client. It's already running for mainstream distros like Ubuntu, but it may need to be explicitly enabled for other distros like Fedora.

Do you think it'd be okay to list this requirement in README.md? If so I'm happy to submit a PR

dorssel, 2024-11-18:
@reynoldsbd 
That sounds more like a "WSL-distro FAQ" for the Wiki. Can you draft something in the Discussions section? I will move it to Wiki then. Thanks!