dorssel/usbipd-win #1020: Not able to connect via USB and send commands using USBIPD because of different IP addresses

Henrique-Liesenfeld-Krever, 2024-08-20:
Hi, i'm trying to use usbipd to connect via usb from my computer to a external machine, it works on linux and macos but not on windows, the problem i've been having is the following.

I have done the procedures described in https://learn.microsoft.com/en-us/windows/wsl/connect-usb when I use the command 

`usbipd attach -w ubuntu --busid 1-8`

I get the following message: 
`usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.27.160.1 to reach the host.`

The problem is, when I type "wsl" on my Powershell, it transitions to the ubuntu subsystem, when using the command:
`hostname -I`
on the wsl cmi i get the following message
"172.27.169.251"
which means, the IP are different, and when i execute my code to connect to my device for some reason (i guess the different IP), my device doesn't get the commands I send him and the code doesnt work properly (it works on linux and macos), is there something i'm missing? If so, please enlighten me, sorry if this is a easy fix and a dumb question, I've been banging my head all day trying to figure this out.

If more information is necessary, please let me know!

dorssel, 2024-08-20:
> Using IP address 172.27.160.1 to reach the host.

This is the *host* IP address, that will be used *from* WSL to reach the Windows host. It is not the IP address of WSL (which is what you got with `hostname -I`).

Is there any error after the attach? What is the output of `lsusb` on WSL after the attach?

dorssel, 2024-08-21:
The device is attached correctly. See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting if you want to see any captures. Looks like any problem is client side (WSL).