dorssel/usbipd-win #1034: could not connect to port 3240：system error

coffee-xzm, 2024-09-19:
Hi,
The error message I am getting is:
`usbip: error: could not connect to 8.8.8.8:3240: System error`
when running `sudo usbip  list --remote=$(cat /etc/resolv.conf | grep nameserver | awk '{print $2}') `
according to [(https://github.com/phuoctan4141/WSL/blob/main/Connect%20USB%20devices/USB%20Camera.md)]
and debug messages ended as follows:
`usbip: debug: names.c:356:[parse] line 24041 class ef Miscellaneous Device
usbip: debug: names.c:409:[parse] line 24042 subclass ef:01 ?
usbip: debug: names.c:442:[parse] line 24043 protocol ef:01:01 Microsoft ActiveSync
usbip: debug: names.c:442:[parse] line 24044 protocol ef:01:02 Palm Sync
usbip: debug: names.c:409:[parse] line 24045 subclass ef:02 ?
usbip: debug: names.c:442:[parse] line 24046 protocol ef:02:01 Interface Association
usbip: debug: names.c:442:[parse] line 24047 protocol ef:02:02 Wire Adapter Multifunction Peripheral
usbip: debug: names.c:409:[parse] line 24048 subclass ef:03 ?
usbip: debug: names.c:442:[parse] line 24049 protocol ef:03:01 Cable Based Association
usbip: debug: names.c:409:[parse] line 24050 subclass ef:05 USB3 Vision
usbip: debug: names.c:356:[parse] line 24051 class fe Application Specific Interface
usbip: debug: names.c:409:[parse] line 24052 subclass fe:01 Device Firmware Update
usbip: debug: names.c:409:[parse] line 24053 subclass fe:02 IRDA Bridge
usbip: debug: names.c:409:[parse] line 24054 subclass fe:03 Test and Measurement
usbip: debug: names.c:442:[parse] line 24055 protocol fe:03:01 TMC
usbip: debug: names.c:442:[parse] line 24056 protocol fe:03:02 USB488
usbip: debug: names.c:356:[parse] line 24057 class ff Vendor Specific Class
usbip: debug: names.c:409:[parse] line 24058 subclass ff:ff Vendor Specific Subclass
usbip: debug: names.c:442:[parse] line 24059 protocol ff:ff:ff Vendor Specific Protocol
usbip: error: could not connect to 8.8.8.8:3240: System error`
I'm a beginner and I have no idea  how to track bugs.
Can you please advise on how to fix?And please inform me of my improper post.
Thanks

dorssel, 2024-09-19:
8.8.8.8 is your nameserver; it is, in fact, a Google DNS server. So, you are trying to attach an USB device from them. I'm quite sure they do not offer USBIP services. You need the IP address of your host that physically has the USB connected to it.

coffee-xzm, 2024-09-20:
> 8.8.8.8 is your nameserver; it is, in fact, a Google DNS server. So, you are trying to attach an USB device from them. I'm quite sure they do not offer USBIP services. You need the IP address of your host that physically has the USB connected to it.

your sincere advice helped me a lot, thanks

closed 2024-09-22 by dorssel