dorssel/usbipd-win #1035: WSL usbip: error: Attach Request for 1-2 failed - Unknown Op Common Status

xyf2002, 2024-09-20:
---

**Description:**

I encountered an error while trying to attach a USB device to my WSL 2 distribution using `usbipd`. The command fails with an "Unknown Op Common Status" error.

**Steps to Reproduce:**
1. Connect the USB device to the Windows host.
2. Run the following command in PowerShell:
   ```powershell
   usbipd wsl list
   ```
   - The device is listed with `busid=1-2`.
3. Attempt to attach the device to the WSL distribution:
   ```powershell
   usbipd attach --wsl --busid=1-2
   ```
4. The following error occurs:
   ```
   usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
   usbipd: info: Using IP address 172.29.48.1 to reach the host.
   WSL usbip: error: Attach Request for 1-2 failed - Unknown Op Common Status
   usbipd: error: Failed to attach device with busid '1-2'.
   ```

**Expected Behavior:**
The USB device should be successfully attached and available in the WSL 2 environment.

**Actual Behavior:**
The attachment fails with an "Unknown Op Common Status" error.

**System Information:**
- **OS:** Windows 11 (Specify version and build number)
- **WSL Version:** WSL 2 Prerelease
- **WSL Distribution:** Ubuntu 22.04
- **usbipd-win Version:** 4.3.0
- **USB Device:** Jlink

**Additional Information:**
- I have verified that the device is recognized correctly in Windows Device Manager.
- I have tried restarting the `usbipd` service and rebooting the system, but the issue persists.
- I ran `usbipd` in debug mode, and the following logs were generated: (Attach any relevant logs if available)
- Network connectivity between WSL and the host is working correctly. I can ping `172.29.48.1` from within WSL.

**Logs and Debug Output:**
(Include any relevant logs from running `usbipd` in debug mode or `dmesg` output from WSL, if applicable)

**Questions:**
- Are there any known limitations or additional configurations required for this type of USB device?
- Could this issue be related to specific WSL 2 kernel configurations or compatibility issues?

**Attachments:**
(Attach any screenshots, logs, or configuration files that might be helpful for diagnosing the issue)

dorssel, 2024-09-20:
> WSL usbip: error: Attach Request for 1-2 failed - Unknown Op Common Status

This is an error from the WSL-side usbip tool. The "Unknown Op Common Status" suggests that the TCP connection fails in a very early stage (even before it connects to the host `usbipd` service). To confirm, can you run the server in a console with debug output? See: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. Also, can you ping the host from within WSL with `ping 172.29.48.1`? (Note that the IP address may have changed by now, it is kind of random every time WSL restarts).

oliver226, 2024-10-11:
Same error in my side.  And I run the usbipd server in a console. But I will easily have below error in wsl ubuntu. If I use run ubuntu in vmware.  Don't see any error like below.
Error: Timed out handling USB events in mpsse_flush().
Error: ftdi device did not return all data: 0, expected 6

oliver226, 2024-10-11:
> Same error in my side. And I run the usbipd server in a console. But I will easily have below error in wsl ubuntu. If I use run ubuntu in vmware. Don't see any error like below. Error: Timed out handling USB events in mpsse_flush(). Error: ftdi device did not return all data: 0, expected 6

My issue solved.
create  .wslconfig file in location :   C:\user\your_account
contents:
# Settings apply across all Linux distros running on WSL 2
[wsl2]
# Disable page reporting so WSL retains all allocated memory claimed from Windows and releases none back when free
pageReporting=false

# Turn on default connection to bind WSL 2 localhost to Windows localhost. Setting is ignored when networkingMode=mirrored
localhostforwarding=true

# Disables nested virtualization
nestedVirtualization=false

# Turns on output console showing contents of dmesg when opening a WSL 2 distro for debugging
debugConsole=true

# Enable experimental features
[experimental]
sparseVhd=true

oliver226, 2024-10-17:
> My issue solved. create .wslconfig file in location : C:\user\your_account contents:
>
> # Settings apply across all Linux distros running on WSL 2
> [wsl2]
>
> # Disable page reporting so WSL retains all [...]

It's weird . not solved. The error will happend again.

xyf2002, 2024-11-10:
> This is an error from the WSL-side usbip tool. The "Unknown Op Common Status" suggests that the TCP connection fails in a very early stage (even before it connects to the host `usbipd` service). To [...]

PING 172.29.48.1 (172.29.48.1) 56(84) bytes of data.
^C
--- 172.29.48.1 ping statistics ---
10 packets transmitted, 0 received, 100% packet loss, time 9371ms   Cannot ping  172.29.48.1

xyf2002, 2024-11-10:
> This is an error from the WSL-side usbip tool. The "Unknown Op Common Status" suggests that the TCP connection fails in a very early stage (even before it connects to the host `usbipd` service). To [...]

This is the log info, sudo usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      4.3.0-70+Branch.master.Sha.701757afaea0ed252dd541f290decafe807970a5
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\13862
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started  ,   stuck here and do not continue

dorssel, 2024-11-11:
> 10 packets transmitted, 0 received, 100% packet loss, time 9371ms Cannot ping 172.29.48.1

There is your problem. WSL is not even able to ping the host. usbipd-win relies on a working network connection.

> Hosting started , stuck here and do not continue

It's not stuck, it is waiting for an incoming connection. But something is blocking connections between WSL and the host. The connection attempt is not reaching the application level (just as it is not for ping).

therds, 2025-01-12:
I stop service  iphlpsvc (IP Helper) is work. IP Helper and usbipd use tpc port 3240.
netstat -an | findstr LISTEN |grep 3240
  TCP    0.0.0.0:3240           0.0.0.0:0              LISTENING  
  TCP    0.0.0.0:3240           0.0.0.0:0              LISTENING

closed 2025-01-14 by xyf2002