dorssel/usbipd-win #1038: usbipd: error: The VBoxUsbMon driver is currently not running; a reboot should fix that.

Zazmi2, 2024-09-26:
Hi I am getting the error: usbipd: error: The VBoxUsbMon driver is currently not running; a reboot should fix that. I have tried rebooting my system, uninstalling and reinstalling usbipd. I am able to bind the device but not attach when doing this: usbipd attach --wsl --busid 1-4 and I get no output when running sc query VBoxUsMon.

I am using a surface pro laptop Windows 11 arm64 processor and I am using WSl2.

[image]

dorssel, 2024-09-26:
There is something wrong with your Windows service manager: either it should give you a status about VBoxUsbMon (whether it is running or not), or it should complain that the service is not installed at all. The fact that it does not say anything is problematic.

I do remember that someone had similar problems where it turned out to be that another 'sc' executable was in the PATH.

Try running explicitly the Windows service manager as:
```
. $env:windir\system32\sc.exe query VBoxUsbMon
```

Zazmi2, 2024-09-27:
Hi, this is what I get: 
[image]

Lewiscowles1986, 2024-09-28:
Okay, so I am seeing this using windows 11 arm64. I wonder if others are perhaps using a different architecture?

```
SERVICE_NAME: VBoxUsbMon
        TYPE               : 1  KERNEL_DRIVER
        STATE              : 1  STOPPED
        WIN32_EXIT_CODE    : 1275  (0x4fb)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
```

Lewiscowles1986, 2024-09-28:
my error when trying to start the service is also different

```
PS C:\Windows\system32> Start-Service VBoxUsbMon
Start-Service : Service 'VirtualBox USB Monitor Service (VBoxUsbMon)' cannot be started due to the following error:
Cannot start service VBoxUsbMon on computer '.'.
At line:1 char:1
+ Start-Service VBoxUsbMon
+ ~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OpenError: (System.ServiceProcess.ServiceController:ServiceController) [Start-Service],
   ServiceCommandException
    + FullyQualifiedErrorId : CouldNotStartService,Microsoft.PowerShell.Commands.StartServiceCommand

```

Lewiscowles1986, 2024-09-28:
According to https://github.com/dorssel/usbipd-win/issues/1023#issuecomment-2310041856 this is win64 only, as it uses VirtualBox components. If true, then I've got my answer.

dorssel, 2024-09-29:
> 

From https://learn.microsoft.com/en-us/windows/win32/debug/system-error-codes--1000-1299-

```
ERROR_DRIVER_BLOCKED

1275 (0x4FB)

This driver has been blocked from loading.
```

Lewiscowles1986, 2024-09-29:
Sorry, are you saying it should load and work, on arm64, or that it still should not?

dorssel, 2024-09-30:
@Lewiscowles1986 

From the README, very first line:

> This software requires Microsoft Windows 10 (x64 only)

From WSL documentation (https://learn.microsoft.com/en-us/windows/wsl/connect-usb), second prerequisite:

> A machine with an x64 processor is required. (x86 and Arm64 are currently not supported with usbipd-win).

From any search in the issue list about ARM64

https://github.com/dorssel/usbipd-win/issues?q=is%3Aissue+arm64

Lewiscowles1986, 2024-10-01:
Right, except I had to find your github, which was linked via the releases page, not your README; from the microsoft page, where the text is NOT prominently displayed. In fact I wasn't forced to read it and it wasn't called out. Don't feel bad, I've found other software that seems to grant me similar https://github.com/vadimgrn/usbip-win2 and does work with ARM

Thank you for creating what you have created, you have no obligation to help anyone, but I am pointing out that this is most unhelpful, your responses have been less than clear. A Simple No would have also done just as well as the above 2 comments.

Lewiscowles1986, 2024-10-01:
[image]
[image]

dorssel, 2024-10-01:
@Zazmi2
OP, are you also using ARM64?

Zazmi2, 2024-10-01:
Yep using arm64

closed 2024-10-02 by dorssel

PURESUN-OYLY, 2024-11-01:
Anyone fix this issue? I also use ARM64 CPU

PURESUN-OYLY, 2024-11-18:
> > Right, except I had to find your github, which was linked via the releases page, not your README; from the microsoft page, where the text is NOT prominently displayed. In fact I wasn't forced to read it and it wasn't called out. Don't feel bad, I've found other software that seems to grant me similar https://github.com/vadimgrn/usbip-win2 and does work with ARM
> > Thank you for creating what you have created, you have no obligation to help anyone, but I am pointing out that this is most unhelpful, your responses have been less than clear. A Simple No would have also done just as well as the above 2 comments.
> 
> How did you use this to forward usb device windows (WOA)-> Linux wsl @Lewiscowles1986 ? from my understanding USBIP-win2 is a USB/IP client for Windows that allows Windows to connect to USB devices shared from Linux servers. It does not support sharing Windows USB devices to WSL?
> 
> Did you manage to get it to work for that purpose? forward device from host to wsl?

I did not find anyway use usbipd on wsl2, i changed my wsl to version 1, then used uart by ttyUARTx. And i have been tested that usbipd can work well on wsl2 with x86 CPU

Lewiscowles1986, 2025-03-26:
Oh I love this. I was just last week setting up virtualbox on my apple silicon, and now it seems that work is also benefitting windows on ARM ❤