dorssel/usbipd-win #1045: usbipd: warning: A firewall appears to be blocking the connection; ensure TCP port 3240 is allowed.

Matszwe02, 2024-10-22:
Tried disabling every firewall restriction, rising privileges to both usbipd-win and usbipd for wsl2, didn't work.
Any help will be appretiated

I basically followed this instruction during install:
https://www.xda-developers.com/wsl-connect-usb-devices-windows-11/

```cmd
>usbipd attach --wsl --busid 8-3
usbipd: info: Using WSL distribution 'Ubuntu-22.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.26.32.1 to reach the host.
usbipd: warning: A firewall appears to be blocking the connection; ensure TCP port 3240 is allowed.
```
```cmd
>wsl --version
WSL version: 2.3.24.0
Kernel version: 5.15.153.1-2
WSLg version: 1.0.65
MSRDC version: 1.2.5620
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
Windows version: 10.0.22631.4169
```
```sh
~$ sudo !!
sudo usbipd
[sudo] password for pi:
usbipd: info: starting usbipd (usbip-utils 2.0)
usbipd: info: listening on 0.0.0.0:3240
usbipd: info: listening on :::3240
^Cusbipd: info: shutting down usbipd
```
```sh
~$ uname -a
Linux DESKTOP-xxxx 5.15.153.1-microsoft-standard-WSL2 #1 SMP Fri Mar 29 23:14:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

dorssel, 2024-10-23:
Could be many things. Have a look through other (possibly closed) issues. Could be 3rd party security software, domain policy, etc.

Matszwe02, 2024-10-23:
> Could be many things. Have a look through other (possibly closed) issues. Could be 3rd party security software, domain policy, etc.

Everything I know is disabled, I've looked at (I think) all of the closed issues

Matszwe02, 2024-10-23:
> What happens if you do `nc 172.26.32.1 3240` from within WSL? (NOTE: the IP address may change every time you restart WSL)

It doesn't respond to that command

dorssel, 2024-10-23:
Then you *do* have connection problems from WSL to the host. That's outside the domain of usbipd-win. Maybe ask for help at https://github.com/microsoft/WSL/issues.

Matszwe02, 2024-10-24:
Thanks for the help!

closed 2024-10-24 by Matszwe02

beyarkay, 2025-02-05:
For future travellers:

> Create or edit %UserProfile%\.wslconfig and add the following:
> ```
> [wsl2]
> networkingMode=mirrored
> ```
> 
> Exit your WSL distribution
> 
> Run wsl --shutdown from PowerShell then restart WSL

bjarkegosvig, 2025-05-19:
Note for further me, as an alternative solution to mirrored network mode:

Windows 11 security policies define three types of network connections by default: “Not configured”, “Private”, and “Public”. Some networks, including home WiFi or public hotspots, may automatically be classified as “Not configured” on a company laptops. Under this classification, none of the predefined firewall rules are applied — meaning all TCP requests targeting the PC's NIC IP are denied. As a result, WSL cannot access the PC's local TCP port 3240.

A local solution is to do: 
Start->Local Security Policy->Network List Managers Policies
and change the "Location Type" from "Not configured" to "Private" for the "Identifying Networks Properties" policy and reboot the firewall issue is gone.

CrisNevares, 2025-07-01:
### For future travelers!

I was able to resolve my issue by adding the `--host-ip` flag and providing the value from ``echo $(host `hostname --long` | grep -oP '(\s)\d+(\.\d+){3}' | tail -1 | awk '{ print $NF }' | tr -d '\r')`` (taken from [here](https://superuser.com/questions/1732399/cannot-communicate-with-windows-localhost-from-wsl2/1732513#1732513)).

Below is my startup script to auto-attach my device (Yubikey) if connected:

```
#! /bin/bash

USBIPD_STATE=$(usbipd.exe state)

CONNECTED_DEVICE=$(echo $USBIPD_STATE | jq -r '.Devices[] | select(.ClientIPAddress != null)')

HOST_IP=$(host `hostname --long` | grep -oP '(\s)\d+(\.\d+){3}' | tail -1 | awk '{ print $NF }' | tr -d '\r')

if [ "$CONNECTED_DEVICE" ]
then
    echo "Device connected"
else
    PERSISTED_CONNECTED_DEVICE=$(echo $USBIPD_STATE | jq -r '.Devices[] | select(.PersistedGuid != null and .BusId != null)')
    if [ "$PERSISTED_CONNECTED_DEVICE" ]
    then
        BUSID=$(echo $PERSISTED_CONNECTED_DEVICE | jq -r '.BusId')
        usbipd.exe attach --busid=$BUSID --wsl --host-ip $HOST_IP
    else
        echo "No bound devices connected"
    fi
fi
```