dorssel/usbipd-win #1046: Yubikey with FIDO2 (or FIDO1 U2F) not recognized in non-WSL guest

rbeede, 2024-10-23:
I am attempting to use a Yubikey 5 (and also 4) inside an Ubuntu 22 Hyper-V guest (not WSL). I specifically want WebAuthN FIDO2 mode in a Chrome/Firefox browser.

I have been successful in sharing from the host and attaching to the USB in the guest, but it appears the necessary USB communication for FIDO2/U2F is not working. This issue seems a little similar to https://github.com/dorssel/usbipd-win/issues/97, but I'm not trying to get OTP (/dev/console) functionality.

From host hypervisor
```
usbipd bind --busid=3-2
```

From the Ubuntu 22 guest VM in Hyper-V
```shell
usbip list --remote=192.168.1.200
Exportable USB devices
======================
 - 192.168.1.200
        3-2: Yubico.com : Yubikey 4/5 OTP+U2F (1050:0403)
           : USB\VID_1050&PID_0403\6&16D50F4A&0&2
           : (Defined at Interface level) (00/00/00)
           :  0 - Human Interface Device / Boot Interface Subclass / Keyboard (03/01/01)
           :  1 - Human Interface Device / No Subclass / None (03/00/00)

sudo usbip attach --remote=192.168.1.200 --busid=3-2

lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 1050:0403 Yubico.com Yubikey 4/5 OTP+U2F
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Yubikey Manager GUI on the guest does see the Yubikey, but trying to auth process with Chrome/Firefox just fails.  Doing a WebAuthN FIDO2 (or fido1 u2f) press on the host hypervisor works fine (when not attached to the guest).

dorssel, 2024-10-23:
Maybe an additional driver is required in the WSL kernel? Are there any errors logged when you do: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting? Anything in WSL syslog?

rbeede, 2024-10-24:
To clarify, I am not using WSL but an independent Ubuntu 22 guest VM running in Hyper-V.

No unexpected errors in the Application event viewer logs. I'll dig into other Ubuntu kernel drivers that may be needed for Yubikeys.

rbeede, 2024-10-26:
No luck with `sudo modprobe vhci-hcd` although `lsusb` does show the Yubikey.

rbeede, 2024-10-26:
I figured out the issue on why web browser could not access the device. When I ran the following it worked in Firefox:

```shell
sudo -i

cp -a /home/rbeede/.Xauthority .

chown root: .Xauthority

# in the sudo root shell still

XAUTHORITY=/root/.Xauthority sudo firefox
```

Visiting https://demo.yubico.com/webauthn-technical/registration my Yubikey was detected immediately.

rbeede, 2024-10-26:
This page hints about the permission issue, and following the instructions worked!

https://developers.yubico.com/yubikey-manager/Device_Permissions.html

rbeede, 2024-10-26:
You may want to add to documentation some of these tips:

On an Ubuntu guest VM the following commands may be needed:
```shell
sudo modprobe vhci-hcd

sudo rmmod usbhid && sudo modprobe usbhid
```

For attaching a Yubikey FIDO2 to an Ubuntu guest VM (non-WSL kernel):
1. https://developers.yubico.com/yubikey-manager/Device_Permissions.html
1. Download the FIDO udev rules
    - 70-u2f.rules
1. `sudo cp 70-u2f.rules /usr/lib/udev/rules.d/`
1. `sudo chmod 0644 /usr/lib/udev/rules.d/70-u2f.rules`
1. `sudo udevadm control --reload-rules && sudo udevadm trigger`
1. Remove and reattach the Yubikey USB (usbip attach)
1. You may need to clear your web browser cache

dorssel, 2024-10-26:
Thanks for finding this out. People will be able to find this in the issues list. Maybe (just maybe) I'll add it to the regular documentation. For now, I'll close this issue.

closed 2024-10-26 by dorssel

LouisWayne, 2026-05-06:
Thank you @rbeede - I fixed the issue with your help.

For other frens, check out the below commands:

```sh
cd /tmp

curl -L -o 70-u2f.rules https://raw.githubusercontent.com/Yubico/libu2f-host/master/70-u2f.rules

sudo cp 70-u2f.rules /etc/udev/rules.d/
sudo chmod 0644 /etc/udev/rules.d/70-u2f.rules

sudo udevadm control --reload-rules
sudo udevadm trigger

# FIDO2 should work at this point. If not, detach/attach again:
sudo usbip detach --port 00
sudo usbip attach --remote 1.2.3.4 --busid 0-0
```