dorssel/usbipd-win #1065: Device Not Mounted when Attached to WSL

jerryfleurival, 2024-11-20:
### _I see it created a directory under media folder called USB, But don't see drive directory of my USB device._

**_Should I see it in the list when I type command "lsblk" into bash shell?_**

dorssel, 2025-01-27:
You should see it with `lsusb`. Did you enable USB storage drivers in the WSL kernel?

dorssel, 2025-02-20:
This means it is not attached yet. Did you run `usbipd attach --wsl ...` ? What was the output?

jerryfleurival, 2025-02-20:
I forgot about that. Now I see my device listed. But, I'm trying to still figure out how to go into the device.

Output:
[image]

dorssel, 2025-02-21:
Do you have a WSL kernel with USB storage drivers enabled?

jerryfleurival, 2025-02-21:
Yes, my WSL Kernel Version is 5.15.167.4-microsoft-standard-WSL2 and I am operating under Windows 10 
Version 10.0.19045 Build 19045.

Here is some Kernel Logs:
[image]

----------------------------------------------------------------------
**Notes:**
Not listed when I type lsblk command.

[image]

WSL must not recognize it as a block device.

dorssel, 2025-02-21:
Your kernel does *not* have a USB storage driver (the default WSL kernel doesn't have that). You either need to compile your own kernel (see https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers) or use the pre-release version of WSL (Windows 11 only), which has it as a module.

jerryfleurival, 2025-02-23:
_I am having issues with branching into linux-msft-wsl-5.15.167.4 and this might take awhile than I expect to figure out._

Output:

[image]

dorssel, 2025-02-23:
Yep, that is a known issue. You cannot checkout the Linux kernel on an NTFS (Windows) file system. The "aux" filename (extension or not) is special. See for example https://github.com/raspberrypi/linux/issues/1254. People have complained to both Microsoft and Linux ... nothing is done about it. You need to checkout in a VM, to a non-Windows mounted file system.

bugparty, 2025-02-26:
Don't lose hope, I wrote an tutorial and it is not complex, just take 10 minutes without any sideeffects.
see here: https://www.linkedin.com/pulse/wsl2-sd-card-support-fix-guide-bowen-han-34x3c/

jerryfleurival, 2025-03-05:
@dorssel Does this mean I have to reinstall usbipd in VM for my Ubuntu OS.

dorssel, 2025-03-06:
No, `usbipd` is the server (host) side, which is Windows. You don't need to install anything in Ubuntu. You need to *replace* the WSL kernel, either with some pre-built kernel (as suggested above), or compile it yourself (also suggested above).

jerryfleurival, 2025-03-06:
@dorssel **My current status:
           _ sdc1 is my USB storage device mounted on media folder._**

[image]

[image]

I am using my other computer which has more resources to handle VM (VirtualBox) and OS is Windows 11.
I notice I have not use `attach `and `bind.`  I'm thinking that is for interacting with remote host or you can say working with two or more computer systems. Is that so?

jerryfleurival, 2025-03-06:
@dorssel Okay. [Oracle VirtualBox](https://www.virtualbox.org/) gave me the GUI to Share and Attach USB device locally. So, I can confirm this  with `usbip` command like in my previous comment and be able to do other tasks like `usbipd`. 

 I have confirmed the use of `attach`,` detach`, and` bind` in VM to reach remote host.  Yes, for reaching the host, you first need `usbipd` running in Windows. I need --remote=<IPv4 address> --busid=<id> with of course admin powers to achieve this. It has automatically mounted inside Media directory. Now I can interact with the files and folders of my USB drive. 

Last, as you see in the screenshot above, I am using pre-built Kernel (Generic Kernel) under Ubuntu, which is fine I guess. But, since the PC I'm using is Windows 11, I could go along with using modules by following your steps in [WSL Support wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers). Either way, I now know I have options. 

Let me know if I'm missing something or if something you like to add as a note. Thanks a lot!

closed 2025-03-06 by jerryfleurival