dorssel/usbipd-win #1066: usbipd - Kvaser Leaft Light v2 -62 error in WSL2

borjamunozf, 2024-11-21:
Hello! I'm trying to attach a Kvaser Leaft Light V2 to WSL2 Ubuntu 22, but I'm facing some problems:

1) Share device in admin mode / bind

```
usbipd list
```

[image]

2) Try to attach & logs from usbipd server shows an issue -62
```usbipd attacah

![imagen](https://github.com/user-attachments/assets/e258b8d3-46e2-4182-964a-2237b4dea903)

3) dmesg

![imagen](https://github.com/user-attachments/assets/60f8669b-e16d-4221-a84b-c83232db95d1)

4) The custom kernel has module & usb support

![imagen](https://github.com/user-attachments/assets/ceae7a00-31e7-4e6d-8f05-f57940299815)

````
cat /proc/config.gz | grep KVASER*
CONFIG_CAN_KVASER_PCIEFD=y
CONFIG_CAN_KVASER_PCI=y
CONFIG_CAN_KVASER_USB=y
```

Any other idea or how to debug further?
Thanks!

dorssel, 2024-11-21:
DNR = Did Not Respond (which translates to ETIME = Timeout on Linux).

See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture to get a PCAP that logs all device communication.

AlBowens, 2026-08-21:
For info, I had exactly the same problem with the same device (Kvaser Leaf Light v2).
When I swapped this device for a Kvaser USBcan Light 4xHS I was able to attach to the device and transmit and receive CAN traffic in WSL2 with no problems.
The issue seems to be specific to this one device.