dorssel/usbipd-win #1067: Unable to use MediaTek M7921U Wi-Fi USB device (re: USBSUP_XFER_STALL -> EPIPE -> -32)  (firmware load failed)

sofakng, 2024-11-21:
I'm trying to use a MediaTek MT7921U Wi-Fi USB device but I'm having a problem.

The device appears in Linux using lsusb:
`Bus 001 Device 005: ID 0e8d:7961 MediaTek Inc. Wireless_Device
`

...but it fails trying to initialize the driver and load the firmware.

Here is a dmesg snippet:
```
[   42.385970] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[   42.514034] usb 1-1: SetAddress Request (2) to port 0
[   42.588933] mt7921u 1-1:1.3: Direct firmware load for mediatek/WIFI_RAM_CODE_MT7961_1.bin failed with error -2
[   42.717985] usb 1-1: reset high-speed USB device number 2 using vhci_hcd
[   42.849999] usb 1-1: SetAddress Request (2) to port 0
[   42.945798] mt7921u 1-1:1.3: Direct firmware load for mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin failed with error -2
[   42.946706] vhci_hcd: unlink->seqnum 348
[   42.946893] vhci_hcd: urb->status -104
[   42.947130] vhci_hcd: unlink->seqnum 349
[   42.947248] vhci_hcd: urb->status -104
[   42.947454] vhci_hcd: unlink->seqnum 350
```

Here is a snippet from the usbipd server with trace logging:
```
dbug: Usbipd.Server[1000]
      new connection from 172.23.180.73
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 212 ms
info: Usbipd.ConnectedClient[1]
      Client 172.23.180.73 claimed device at 16-2 (USB\VID_0E8D&PID_7961\000000000).
dbug: 16-2.0[1000]
      Endpoint created
dbug: 16-2.0[1000]
      Masked USB_CONFIG_REMOTE_WAKEUP
dbug: 16-2.0[1000]
      Masked USB_CONFIG_REMOTE_WAKEUP
dbug: 16-2.0[1000]
      Endpoint created
dbug: 16-2.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 16-2.0[1000]
      Masked USB_CONFIG_REMOTE_WAKEUP
dbug: 16-2.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 16-2.5[1000]
      Endpoint created
dbug: 16-2.4[1000]
      Endpoint created
dbug: 16-2.8[1000]
      Endpoint created
trce: 16-2.4[1001]
      Unlinking: PendingSubmits=128, PendingUnlinks=1
trce: 16-2.4[1001]
      Unlinking: PendingSubmits=128, PendingUnlinks=2
trce: 16-2.4[1001]
      Unlinking: PendingSubmits=128, PendingUnlinks=3
trce: 16-2.4[1001]
      Unlinking: PendingSubmits=128, PendingUnlinks=4
trce: 16-2.4[1001]
      Unlinking: PendingSubmits=128, PendingUnlinks=5

...

      Unlinking: PendingSubmits=128, PendingUnlinks=127
trce: 16-2.4[1001]
      Unlinking: PendingSubmits=128, PendingUnlinks=128
trce: 16-2.4[1001]
      Aborting endpoint
dbug: 16-2.4[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: 16-2.4[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: 16-2.4[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
```

I've attached a full usb pcap log.

Can anybody help me?

[usbip-pcap.zip](https://github.com/user-attachments/files/17848563/usbip-pcap.zip)

sofakng, 2024-11-21:
This appears to be similar to #1022

ashokbqc, 2025-04-09:
Was this issue ever properly resolved? 
I am seeing similar issue (albeit not with Mediatek boards). More specifically, the issue seems to be happening when performing a bulk transfer using libusb.