dorssel/usbipd-win #1076: Unable to attach to WSL on Snapdragon X-elite

NeveIsa, 2024-11-28:
PS C:\Windows\system32> usbipd bind --busid 5-1
usbipd: info: Device with busid '5-1' was already shared.

PS C:\Windows\system32> usbipd attach --wsl --busid 5-1
usbipd: error: The service is currently not running; a reboot should fix that.

*** Reboots don't fix anything ***

System Info 
```
OS Name:                       Microsoft Windows 11 Pro
OS Version:                    10.0.26100 N/A Build 26100
OS Manufacturer:               Microsoft Corporation
OS Configuration:              Standalone Workstation
OS Build Type:                 Multiprocessor Free
Registered Owner:              [REDACTED]
Registered Organization:       N/A
Product ID:                    00355-61664-34203-AAOEM
Original Install Date:         6/25/2024, 10:54:06 PM
System Boot Time:              11/27/2024, 4:19:15 PM
System Manufacturer:           LENOVO
System Model:                  83ED
System Type:                   ARM64-based PC
Processor(s):                  1 Processor(s) Installed.
                               [01]: ARMv8 (64-bit) Family 8 Model 1 Revision 201 Qualcomm Technologies Inc ~1920 Mhz
```

Janis-Albertins, 2024-11-28:
I can confirm. Not working on Lenovo with snapdragon

orbli, 2024-12-05:
Bet this software does not yet have arm support. In readme.md it says

>  Microsoft Windows 10 (x64 only)

However I would like to leave my two cents that surface laptop 7 (win11 arm) does not work either, and looking forward if there is any plan to support.

dorssel, 2024-12-06:
> System Type:                   ARM64-based PC

The VirtualBox drivers (which usbipd-win relies on) are for x64 only.
Duplicate of #486

closed 2024-12-06 by dorssel

dorssel, 2025-03-24:
ARM64 support is part of version 5.0.0, which has just been released.

NeveIsa, 2025-03-24:
Thanks for the update.