dorssel/usbipd-win #1077: use auto-attach without check the device existence

BrokenClient, 2024-11-30:
Hi, I am developing an embedded system with MCU. Thank you for the excellent project that I can use the debug probe conveniently.

Is there any way to use auto-attach without checking the device existence?  When I plug in a probe and run `usbipd attach -wsl -a -b 1-1`, it works, and I can unplug it and use another probe. However, it causes an error when the command is executed without a device plugged in. I am going to add this command to Windows Startup so that I can use the PC USB port as a "Linux USB port" if the command would not report error and exit.  So is there any way to do that?

rfk-magventure, 2024-12-27:
I have the same question/issue!

dorssel, 2024-12-27:
> So is there any way to do that?

Not yet, with the current script. But this sounds like a nice generic feature. I will include it somewhere in the Near Future(tm).

dorssel, 2025-01-27:
Can you try the current master build? It has a new option `--unplugged` to be used together with `--auto-attach`.
You can find the installer here: https://github.com/dorssel/usbipd-win/actions/runs/12995345285

dorssel, 2025-02-05:
Now released with 4.4.0

closed 2025-02-05 by dorssel