dorssel/usbipd-win #1085: Unable to Bind (without --force) with SEGGER Jlink devices

biotto, 2024-12-10:
Hello,

I got an issue recently, I was able to bind without any issue with all my Jlink devices and, after attempting to reinstall usbipd for testing a script (which needs to auto-install usbipd using "winget install "), I can not bind to these Jlinks devices anymore without using the --force argument. 

**Note:** 
I have, by mistake, uninstalled usbipd while it was running, I don't know if this may have screw up my authorisation or anything. I have tried to uninstall and reinstall properly using the .msi installer but nothing helps.

Unfortunately, I did not succeed in logging any information by following the given tutorial.

Can you help me?

Thank you!

MichaelFeistETC, 2025-01-23:
Just adding on to this, me and some of the other people on my team also often have problems with usbipd and SEGGER JLink. Requiring --force binds, failing to bind even with --force, hanging usbipd attach calls, and JLinks not working properly once attached and showing in WSL.

dorssel, 2025-01-24:
Can you get a capture as explained here: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting