dorssel/usbipd-win #1092: Unable to "attach" usb device through WSL (docker desktop / W11)

luizguilhermeg, 2024-12-17:
Hello!

I am getting the following error message when trying to attach an USB device to docker desktop (WSL 2 on W11) after successfully binding.

```
ERROR - usbipd: info: Using WSL distribution 'docker-desktop' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
```

I have tried WLS USB Manager 5.5.0, cmd user (usbipd attach --wsl --busid=1-4), and cmd admin (usbipd attach -w --busid 1-4) to no avail.

Here's my usbipd list for reference.

```
BUSID    DEVICE                                                                       STATE
1-2    USB Input Device                                                            Not shared
1-3    USB-Enhanced-SERIAL CH9102 (COM3)                        Shared
1-4    Silicon Labs CP210x USB to UART Bridge (COM5)          Shared
```

Output of echo %USERNAME% (Command Prompt) matches expected username in both user/admin.

Is there anything that I should try?
Thanks

ValentinTwin1206, 2024-12-18:
Hi @luizguilhermeg, 

the `docker-desktop` WSL2 distro is a special purpose distro primarily designed to run container workloads.
Either integrate Docker Desktop in a fully-featured WSL2 distro or use the USB/IP feature of Docker Desktop with its Hyper-V backend:
https://docs.docker.com/desktop/features/usbip/

Best Valentin

luizguilhermeg, 2024-12-19:
Oh, got it! Thank you very much for the prompt response.

closed 2024-12-19 by luizguilhermeg