dorssel/usbipd-win #1095: Bug: Device not visible under `lsblk` in WSL2 Ubuntu 22.04

thernstig, 2024-12-23:
```sh
# On PowerShell side:
> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
4-3    0951:1666  USB-masslagringsenhet                                         Attached
4-11   0489:e0e2  MediaTek Bluetooth Adapter                                    Not shared
4-14   1e71:300c  NZXT Kraken Elite, USB-enhet                                  Not shared
4-16   0b05:19af  AURA LED Controller, USB-enhet                                Not shared
6-1    05ac:024f  USB-enhet                                                     Not shared
6-3    048d:8295  USB-enhet                                                     Not shared
6-6    0bda:1100  USB-enhet                                                     Not shared

Persisted:
GUID 

# On WSL2 (Ubuntu) side:
> lsblk
NAME
    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda   8:0    0 388.4M  1 disk
sdb   8:16   0     8G  0 disk [SWAP]
sdc   8:32   0     1T  0 disk /mnt/wslg/distro
                              /
> lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2/50 Kyson
```

If I do a `wsl --shutdown` the `usbipd list` command shows the device is `Shared` again (as opposed to `Attached`) so restarting WSL does not work.

I am using the latest Ubuntu and WSL kernel release.

What is wrong?

ValentinTwin1206, 2024-12-23:
Your Linux kernel must have USB_STORAGE enabled. Have you checked this?

dorssel, 2024-12-23:
> Your Linux kernel must have `USB_STORAGE` enabled. Have you checked this?

You can check with running from within WSL:

```
zgrep CONFIG_USB_STORAGE /proc/config.gz
```

thernstig, 2025-01-23:
It was not enabled:

```sh
> zgrep CONFIG_USB_STORAGE /proc/config.gz
# CONFIG_USB_STORAGE is not set
```

Why is this requirement not in the README.md? Should it not?

dorssel, 2025-01-24:
@thernstig 

It is in the README:

> See the [wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support) on how to add drivers for USB devices that are not supported by the default WSL 2 kernel.

There are multiple version of Windows, each with multiple versions of WSL. Some do, and some don't have drivers for certain hardware. This project is not just about WSL. For WSL support, visit https://github.com/microsoft/WSL/issues

thernstig, 2025-01-24:
> It is in the README:

I'd argue that is in the wiki, but to me it seems like a key piece of information for anyone using usbipd-win with WSL2, as it is default off in those scenarios. I agree I could read all docs of course, but something mandatory might be better shown in the README.md directly.

closed 2025-01-24 by thernstig

ValentinTwin1206, 2025-01-24:
You should also consider the different filesystems EXFAT, FAT32 etc that are by default also not supported by the wsl / Linux kernel… I guess it’s more than a hard piece to cover this in a readme, however, this is for what the community is for