dorssel/usbipd-win #1097: bind --force will disconnect immediately after connect for Pixel 6 pro  Android debugging mode.

androidfans, 2024-12-25:
[image]

When i use sudo usbipd bind -i 18d1:4ee7 --force, it will bind. then the usb device will disappear. then i re-plug usb, i hear the connect sound, then disconnect sound immediately.

but bindbing without --force doesn't has this issue

dorssel, 2024-12-25:
Bind force just sets the driver permanently. But only for the current device ID. Maybe your device identifies as different IDs (some do that, after changing their USB configuration). Your screenshot shows Shared. That is from a normal bind. Bind force results in Shared (Forced).

Right after bind force, what does the list show?

androidfans, 2024-12-31:
it seems like device id not changed.
[image]

this issue only happens on my Pixel 6 Pro device, my OnePlus device not has this issue.

BTW, when i trying to get repro screenshot, i found that sometimes bind --force will not take effect.

[image]

androidfans, 2024-12-31:
https://github.com/user-attachments/assets/1c738586-b6f3-4d4a-889c-ca3ea5f915cd

androidfans, 2024-12-31:
i think i have found the root cause,
Step:
1. unbind
2. plug in
3. change mode from charging mode to MTP
4.  wait device id change to 18d1:4ee2
5. bind this new id and attach it
6. worked, not disconnect

closed 2024-12-31 by androidfans