dorssel/usbipd-win #1125: Unable to attach a device to WSL2 with docker running

charleslacour, 2025-01-31:
When trying to attach a USB CH341a programmer to a Ubuntu 24.04 instance of WSL2 I was getting an error that said that there may be a firewall stopping the connection.  There is a windows firewall policy allowing usbipd to communicate and there is no other firewall involved.  

After a lot of troubleshooting, I found that if I shutdown the docker service running under WSL2 I was able to attach the programmer via usbipd.  

By default docker was using a the /24 networks in the 172.17.0.0/16 network and the IP network for WSL2 was 172.20.111.0/24.  As a further test I updated the default IP range that docker uses to a /24 in 172.128.0.0/16 and this allowed usbipd to attach a device while docker was running.  

Not sure what the logic is on choosing the interface/address that the injected client binds to but it seems that it is trying to use the lowest IP on all of the interfaces, even virtual interfaces that cannot communicate outside the WSL2 instance.

dorssel, 2025-01-31:
WSL networking / connectivity has changed quite a bit since the last release. Could you try a current master build and confirm that it works better now?

You can find the build here: https://github.com/dorssel/usbipd-win/actions/runs/13042993673

charleslacour, 2025-01-31:
I was running what was listed as the latest version 4.3.0 but I see that the link you provided is 4.3.1.  After putting my docker networks back to 172.17.0.0/16 and updating to the 4.3.1 version of usbipd it works normally.  

Thanks for the quick response and a wonderful tool.

closed 2025-01-31 by charleslacour