dorssel/usbipd-win #1134: Uninstaller of usbipd-win 4.4.0 hangs

kaplan2539, 2025-02-13:
When trying to uninstall usbipd-win 4.4.0, the uninstaller hangs for several minutes and at some point gives up as it cannot stop the VirtualBox USB Monitor Service (VBoxUSBMon). This happens when uninstalling via "Add or remove programs" in the Windows start menu and also when using uninstalling via msiexec. All this is experienced as a user with local administrator rights on Windows 10 professional.

[image]

From the msiexec logs:
```
MSI (s) (D8:9C) [08:44:12:689]: Executing op: CustomActionSchedule(Action=UnbindAll,ActionType=3137,Source=BinaryData,Target=WixQuietExec,CustomActionData="C:\Program Files\usbipd-win\usbipd.exe" unbind --all)
MSI (s) (D8:D8) [08:44:12:704]: Invoking remote custom action. DLL: C:\WINDOWS\Installer\MSI9564.tmp, Entrypoint: WixQuietExec
MSI (s) (D8:14) [08:44:12:705]: Generating random cookie.
MSI (s) (D8:14) [08:44:12:723]: Created Custom Action Server with PID 2972 (0xB9C).
MSI (s) (D8:24) [08:44:12:779]: Running as a service.
MSI (s) (D8:24) [08:44:12:784]: Hello, I'm your 64bit Elevated Non-remapped custom action server.
MSI (s) (D8:9C) [08:44:12:967]: Executing op: ActionStart(Name=StopServices,Description=Stopping services,Template=Service: [1])
MSI (s) (D8:9C) [08:44:12:967]: Executing op: ProgressTotal(Total=3,Type=1,ByteEquivalent=1300000)
MSI (s) (D8:9C) [08:44:12:968]: Executing op: ServiceControl(,Name=usbipd,Action=2,,)
MSI (s) (D8:9C) [08:44:12:968]: Executing op: ServiceControl(,Name=usbipd-win,Action=2,,)
MSI (s) (D8:9C) [08:44:12:968]: Executing op: ServiceControl(,Name=VBoxUSBMon,Action=2,,)
MSI (s) (D8:9C) [08:48:22:410]: Product: usbipd-win -- Error 1921. Service 'VirtualBox USB Monitor Service' (VBoxUSBMon) could not be stopped. Verify that you have sufficient privileges to stop system services.

Error 1921. Service 'VirtualBox USB Monitor Service' (VBoxUSBMon) could not be stopped. Verify that you have sufficient privileges to stop system services.
MSI (s) (D8:9C) [08:48:22:432]: Note: 1: 2265 2:  3: -2147287035
MSI (s) (D8:9C) [08:48:22:440]: User policy value 'DisableRollback' is 0
MSI (s) (D8:9C) [08:48:22:440]: Machine policy value 'DisableRollback' is 0
Action ended 8:48:22: InstallFinalize. Return value 3.
MSI (s) (D8:9C) [08:48:22:443]: Note: 1: 2318 2:
```

The Service does indeed still exist and is in StopPending state:
```
PS C:\Users\kaplan2539> Get-Service -Name VBoxUSBMon

Status   Name               DisplayName
------   ----               -----------
StopP... VBoxUSBMon         VirtualBox USB Monitor Service

PS C:\Users\kaplan2539> (Get-Service -Name VBoxUSBMon).status
StopPending
```

Attempts to stop the service manually in Powershell fails:
```
PS C:\Users\kaplan2539> Get-Service -Name VBoxUSBMon | Stop-Service -Force
Stop-Service : Service 'VirtualBox USB Monitor Service (VBoxUSBMon)' cannot be stopped due to the
following error: Cannot stop VBoxUSBMon service on computer '.'.
At line:1 char:32
+ Get-Service -Name VBoxUSBMon | Stop-Service -Force
+                                ~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : CloseError: (System.ServiceProcess.ServiceController:ServiceController) [St
   op-Service], ServiceCommandException
    + FullyQualifiedErrorId : CouldNotStopService,Microsoft.PowerShell.Commands.StopServiceCommand
```

Is this an upstream bug in VBoxUSBMon?

kaplan2539, 2025-02-13:
After a reboot it is possible to uninstall usbipd-win 4.4.0 without issues.

dorssel, 2025-02-13:
Did you perhaps have any "forced bound" devices when this first failed?

kaplan2539, 2025-02-13:
Not that I remember - I'll repeat my tests and make sure there are no forced bound devices.
But doesn't the uninstaller unbind all devices? Or does that exclude force bound devices?

kaplan2539, 2025-02-14:
So I was able to reproduce the hanging uninstallation with a force-bound USB device...
Any advice here? Should I `usbip unbind` first and sleep for a while before trying to uninstall?

dorssel, 2025-02-14:
I don't know what the root cause is.

The order of things is (at least by design) correct:
1) stop `usbipd` service
    This will detach any currently actively forwarded devices, and only finish after all devices have been released.
2) unbind all devices
    This will replace any forced bound VBoxUSB drivers with the default driver for those devices (which is never VBoxUSB.sys)
3) uninstall VBoxUSB and VBoxUSBMon

After this, no VBoxUSB drivers should be active, and therefore VBoxUSBMon should be idle.

It happens here: https://github.com/dorssel/usbipd-win/blob/master/Installer/Server.wxs#L119-L122

Your error is: "cannot stop VBoxUSBMon".  From the source code of VBoxUSBMon that should only happen if there is a VBoxUSB driver loaded and registered with VBoxUSBMon. However ... forced bound VBoxUSB drivers are not handled well by VBoxUSBMon. It doesn't expect VBoxUSB drivers to register for devices it didn't filter itself. Maybe that messes up the bookkeeping in VBoxUSBMon, causing it to refuse to unload (essentially a reference counting problem).

kaplan2539, 2025-02-14:
Do I understand correctly, I can first stop the `usbipd` service, and then call `usbipd unbind`?

dorssel, 2025-02-14:
Yes

kaplan2539, 2025-02-18:
I've been testing a bit more:
- as soon as a USB device has been shared, usbipd cannot be uninstalled anymore.
- the same is true even if the usb device is detached and unbound before uninstall attempt.
- if the machine is rebooted and no usb device has been shared since boot, usbipd can be uninstalled.
This is different from usbipd-win_4.2.0.msi. I have not tested usbipd-win_4.3.0.msi.
I don't know if this is a change in how usbipd-win handles VBoxUSBMon or if it is an upstream change/bug/feature in VBoxUSBMon.

kaplan2539, 2025-02-18:
This is probably not the highest priority at the moment (after all installation works and everything is functional), but we might run into troubles with the next version of usbipd-win that updates VBoxUSBMon... if users have been sharing a USB device before. Workaround of course is to unbind all, reboot and then install the future usbip-win version ;-)

dorssel, 2025-02-18:
It's indeed what I expected. As soon as you force-bind a device, the VBoxUSB driver messes up the VBoxUSBMon bookkeeping. It will log an entry in Windows EventLog:
```
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="VBoxUSBMon" /> 
    <EventID Qualifiers="49156">4</EventID> 
    <Version>0</Version> 
    <Level>2</Level> 
    <Task>0</Task> 
    <Opcode>0</Opcode> 
    <Keywords>0x80000000000000</Keywords> 
    <TimeCreated SystemTime="2025-02-18T22:43:51.1176877Z" /> 
    <EventRecordID>602134</EventRecordID> 
    <Correlation /> 
    <Execution ProcessID="4" ThreadID="19892" /> 
    <Channel>System</Channel> 
    <Computer>Oakdale</Computer> 
  <Security /> 
  </System>
  <EventData>
    <Data>\Device\VBoxUSBMon</Data>
 
 <Binary>00001D000100000000000000040004C0020000000000000000000000000000000000000000000000494E5445524E414C5F494F43544C5F50524F58595F5354415254555000</Binary> 
  </EventData>
</Event>
```

After that, it refuses to stop/unload, even if the device is unbound again. It will simply remain in the STOP_PENDING state. Not something that we can fix; a reboot will always be required. We could ask Oracle to "fix" the VBoxUSBMon driver such that it handles our use case (forced loading of VBoxUSB without VBoxUSBMon triggering).

dorssel, 2025-03-01:
Thanks for the report!

It has helped identifying a regression bug that was fixed in https://github.com/dorssel/usbipd-win/pull/1153. I have also added an additional test in https://github.com/dorssel/usbipd-win/pull/1155 to prevent future mishaps. I will release a new version soon that should fix this issue.

kaplan2539, 2025-03-03:
Thank you for creating this project and maintaining it so well! I'm happy my report was helpful.

closed 2025-03-03 by kaplan2539

dorssel, 2025-03-12:
Should be fixed now in 4.4.1: https://github.com/dorssel/usbipd-win/releases/tag/v4.4.1