dorssel/usbipd-win #1144: usbipd: error: Failed to attach device with busid '1-9'.

bujtor-nokia, 2025-02-21:
Hi, I am unable to attach a device to wsl. I performed the following commands:

1. usbipd list
2. usbipd attach --wsl --force --busid 1-9
3. usbipd attach --wsl --busid 1-9

The error that I get is:

usbipd attach --wsl --busid 1-9
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Detected networking mode 'nat'.
usbipd: info: Using IP address 172.27.32.1 to reach the host.
WSL usbip: error: Attach Request for 1-9 failed - Device in error state
usbipd: error: Failed to attach device with busid '1-9'

The logs from the usbipd server are:

fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) + 0x153
         at Usbipd.VBoxUsb.<ClaimDeviceOnce>d__0.MoveNext() + 0x6b
      --- End of stack trace from previous location ---
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x17e
      --- End of stack trace from previous location ---
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x2b9
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0xf03
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0x18f9
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x32f
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed

The things that I have tried to fix:

- reinstall usbipd
- reinstall the driver of the device
- resintall wsl
- I checked on 2 other identical PCs (same windows, same driver versions, same wsl) it worked on both of them

bujtor-nokia, 2025-02-21:
@dorssel Yes, you are right, I modified the original post

dorssel, 2025-02-21:
Some devices cannot handle dynamic driver switching well. What may help is:
```
usbipd bind --force --busid 1-9
<reboot>
usbipd attach --wsl --busid 1-9
```

Remember that binding is persistent; you only have to do it once and it survives reboot. Unbinding or binding again *without* `--force` will actually undo the previous force (and likely will require another reboot, and likely will not work for your device).

dorssel, 2025-02-28:
Can you check in Windows Device Manager and see what the state of the "USBIP Shared Device" is?

jmzagorski, 2025-02-28:
I saw that same error message in Event Viewer after a windows 11 update. I had to unbind all my existing bindings and re-bind. It seems like @bujtor-nokia tried that based on his steps, but I wanted to post this in case someone else needed help.