dorssel/usbipd-win #1145: Attach to wsl2 failed with a specific USB expansion hub.

woniu7, 2025-02-22:
- When I attach to wsl with a USB expansion hub, it failed most of the time while it succeeds occasionally.
- It always succeeds using motherboard USB or another  USB expansion hub.
- Is it a problem with my expansion slot? Or a bug of usbipd?
- Here is my version and logs.

~ $ usbipd --version
`4.3.0-70+Branch.master.Sha.701757afaea0ed252dd541f290decafe807970a5.701757afaea0ed252dd541f290decafe807970a5`
~ $ wsl --version
```
WSL version: 2.4.11.0
Kernel version: 5.15.167.4-1
WSLg version: 1.0.65
MSRDC version: 1.2.5716
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
Windows version: 10.0.19045.5487
```

Attach failed:
```
usbipd: info: Device with hardware-id '1050:0402' found at busid '5-4'.
usbipd: info: Using WSL distribution 'NixOS' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.30.112.1 to reach the host.
WSL usbip: error: Attach Request for 5-4 failed - Device in error state
usbipd: error: Failed to attach device with busid '5-4'.
```

usbipd server error log：
```
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ComponentModel.Win32Exception (2): CreateFile
         at Usbipd.DeviceFile..ctor(String) + 0xdc
         at Usbipd.VBoxUsb.<ClaimDeviceOnce>d__0.MoveNext() + 0x9b
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x145
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0xd99
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0x1823
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x300
```

And sometimes it suddenly succeeds
```
info: Usbipd.ConnectedClient[1]
      Client 172.30.126.149 claimed device at 5-4 (USB\VID_1050&PID_0402\7&2B48A09B&0&4).
```

dorssel, 2025-02-22:
- When it fails, does it take around 10 seconds before the error?
- When it succeeds, how long does it take to attach?
- Can you locate the expansion hub in Device Manager? What are the driver details of the *hub*?

woniu7, 2025-02-22:
> * When it fails, does it take around 10 seconds before the error?
No, it failed quickly.
>     * When it succeeds, how long does it take to attach?
Immediately too.
>     * Can you locate the expansion hub in Device Manager? What are the driver details of the _hub_?
It show me two news in device manager when I plugin the _hub_.

1. Generic SuperSpeed USB Hub
Universal Serial Bus controllers
Port_#0005.Hub_#0001

Driver Version: 10.0.19041.4355
Dever Files: C:\windows\system32\DRIVERS\USBHUB3.SYS
Hardware Ids: USB\VID_05E3&PID_0626&REV_0663

3. Generic USB hub
Driver Version: 10.0.19041.4355
Dever Files: C:\windows\system32\DRIVERS\USBHUB3.SYS
Hardware Ids: USB\VID_05E3&PID_0626&REV_0663

woniu7, 2025-02-22:
Oh， It seems works after Unchecking the option 'Allow the computer to turn off this device to save power' .

dorssel, 2025-02-22:
Is that for the hub, or for the device-to-be-shared itself?

woniu7, 2025-02-23:
> Is that for the hub, or for the device-to-be-shared itself?

The hub.

dorssel, 2025-02-23:
OK, interesting. But not much we can do about. At least you found a way to make it work. Problem solved?

woniu7, 2025-02-24:
> OK, interesting. But not much we can do about. At least you found a way to make it work. Problem solved?

Thanks, communication is the solution！

closed 2025-02-24 by woniu7