dorssel/usbipd-win #1146: How to share usb printer with scanner in both windows

rajputsagar905, 2025-02-22:
Share commands line & guide me

rajputsagar905, 2025-02-22:
Also confirm without installing WSL in both windows pc is possible to share usb printer scanner??

dorssel, 2025-02-22:
You do not need WSL, you can use any client.

rajputsagar905, 2025-02-23:
Command line for attach with shared host pc printer on client pc, pls share commands line for attach device on cmd

rajputsagar905, 2025-02-23:
I shared usb printer successfully but not getting attach with printer from client windows system 

Suggest command line for connect

rajputsagar905, 2025-02-23:
My host pc ip is 192.168.137.1
Bus id 3-2
Suggest command line for attach with same

dorssel, 2025-02-23:
This software is for the server side only. See also the README. You need other software for the client side.