dorssel/usbipd-win #1157: Driver detected an internal error in its data structures for \Device\VBoxUSBMon.

jemp, 2025-03-02:
Love the software.

I'm running a USB Bluetooth doggle on my host machine for my Hyper-V Linux installation.

Seems I randomly get the error 'Driver detected an internal error in its data structures for \Device\VBoxUSBMon.', which will drop all client connections.

Has anything like this been reported before?

Thank you!

dorssel, 2025-03-02:
No, haven't seen this one. I guess this is in Windows EventLog? can you post the full event data?

jemp, 2025-03-02:
> No, haven't seen this one. I guess this is in Windows EventLog? can you post the full event data?

`<Events>
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
<Provider Name="VBoxUSBMon"/>
<EventID Qualifiers="49156">4</EventID>
<Version>0</Version>
<Level>2</Level>
<Task>0</Task>
<Opcode>0</Opcode>
<Keywords>0x80000000000000</Keywords>
<TimeCreated SystemTime="2025-03-02T14:31:24.0035667Z"/>
<EventRecordID>47463</EventRecordID>
<Correlation/>
<Execution ProcessID="4" ThreadID="1660"/>
<Channel>System</Channel>
<Computer>HMS-Automation</Computer>
<Security/>
</System>
<EventData>
<Data>\Device\VBoxUSBMon</Data>
<Binary>00001D000100000000000000040004C0020000000000000000000000000000000000000000000000494E5445524E414C5F494F43544C5F50524F58595F5354415254555000</Binary>
</EventData>
<RenderingInfo Culture="en-US">
<Message>Driver detected an internal error in its data structures for \Device\VBoxUSBMon.</Message>
<Level>Error</Level>
<Task>None</Task>
<Keywords>
<Keyword>Classic</Keyword>
</Keywords>
</RenderingInfo>
</Event>
</Events>`

It could be faulty hardware, bit I wanted to rule out anything software related.

dorssel, 2025-03-03:
O, I am sorry. I *have* seen this before. It is actually "normal" when you use `bind --force`. It is benign (if you indeed used `bind --force`).

jemp, 2025-03-03:
> O, I am sorry. I _have_ seen this before. It is actually "normal" when you use `bind --force`. It is benign (if you indeed used `bind --force`).

Thank you, I'm going to close this. My issue seems to be external to USBIPD.

closed 2025-03-03 by jemp