dorssel/usbipd-win #1160: `--force` has no effect on `bind`

thanhminh-vht, 2025-03-10:
## Description
Running `usbipd bind --force` still result in `"IsForced": false`.

## Reproduce
1. `usbipd bind -f -i 1234:ABCD`
2. `usbipd state`

## Expected result
The binded device should have `"IsForced": true`.

## Actual result
The binded device has `"IsForced": false`.

## Additional information
usbipd v4.4.0
I tried everything that I could think of: Restarting my computer, unplug and plug, uninstall driver, etc. So far nothing worked.

dorssel, 2025-03-10:
Yes, I found this regression bug last week. It has been fixed in master.

`binf --force` does take effect. But it is detected/reported wrong (see fix #1153).

You can find the (unsigned) installer with the fix here: https://github.com/dorssel/usbipd-win/actions/runs/13750435072, or you will have to wait a few days/weeks for the next release.

I recommend that you unplug the device and then reboot before upgrading.

dorssel, 2025-03-12:
Please update to release 4.4.1: https://github.com/dorssel/usbipd-win/releases/tag/v4.4.1

closed 2025-03-12 by dorssel