dorssel/usbipd-win #1167: Not able to add more then 8 usb devices

prakh20, 2025-03-20:
I have around 15 usb devices connected with my windows system, but I am able to share only 8 usb device and getting "no port found" error while adding others.

dorssel, 2025-03-20:
To WSL/Linux I assume?
That's a limitation of the Linux kernel that emulates the hub.
You will have to compile your own with modified parameters: https://elixir.bootlin.com/linux/v6.13.7/source/drivers/usb/usbip/vhci.h#L75-L89

The default is 1 hub USB 2.0/1.1 supporting 8 devices, and 1 hub USB 3.0 supporting 8 devices.
Either increase the number of devices per hub, or increase the number of hubs (or both).

prakh20, 2025-03-22:
Yes I am trying on wsl/linux
Even though I have changed the number of controller to 5 and build a custom kernel, I am able to see only 1 host controller created.