dorssel/usbipd-win #1172: "Unbind or unplug while attached" error while making connection from Hyper-V VM to Hyper-V host

jjanczyk, 2025-03-24:
Hi!

While remotelly supporting my colleague, we faced a strange error today. We were unable to share FTDI converter attached to the Windows host running Hyper-V VM with Arch Linux. The error on Linux was quite generic: "usbip: error: import device", while on Windows we capture the following log:
[image]

The connection was fine as we were able to execute "usbip list -r <host_ip>" on Linux and the shared device was listed there. Moreover the same error was encountered when my colleague tried to connect to my host having the identical USB dongle attached. From the other hand, the same Linux image that I later shared with my colleague was running fine on my laptop.

The only difference I spotted is the configuration version of the Hyper-V VM - it was version 9.0 for my colleague (running on Windows 10), while I had version 11.0 and 12.0 on the two laptops that I used.

Please help me further investigate the issue. I am quite unsure which side of the connection to blame. Linux is suspected as it failed to connect to two different Windows hosts, but from the other hand the same Linux image (same virtual drive mounted to different Hyper-V VM) was working fine on newer Windows.

Any guidance would be highly appreciated.

Thanks,
JJ

dorssel, 2025-03-25:
Sounds like the `vhci_hcd` module wasn't loaded, which would lead to these symptoms. Try `sudo modprobe vhci_hcd` before doing the attach.

jjanczyk, 2025-03-25:
Hi!
Thanks for reviewing this. However, the module was loaded. When I forgot to do that for the first time, Linux explicitly stated that prompting me to load the driver what I did. I saw in the dmesg log that the driver was loaded (I believe) correctly.

dorssel, 2025-03-25:
Can you make a USB capture. Just to see if there is any USB traffic at all, or that things are already going wrong at the networking level. See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture

jjanczyk, 2025-03-26:
Hi Frans,
Sorry for taking your time. Problem solved. I've just found that I'm clumsy and that's all the story...

I was switching the environments including hoping between WSL and "full" Hyper-V and I missed the clue. I was running the usbip from the regular user and the error message was a bit missleading. It run like a charm when I executed it from the root account. I have had to do it this way before but simply forgot abut this "small" detail...

closed 2025-03-26 by jjanczyk