dorssel/usbipd-win #1176: attach doesn't work with only docker-desktop distro installed

andrewleech, 2025-04-02:
Seen on a computer where WSL was only being used to run docker containers, when trying to use usbipd to attach a device the following error is seen `WSL env: can't execute 'bash': No such file or directory`

[image]

Turns out docker desktop on windows (version 4.39.0 in this case) has such a cut down wsl distro running it does not even include bash.

This is with usbipd-win 5.0.0-10+Branch.master.Sha.7a8c00f5603c85cfce48f0c439c7c90763f1e105.7a8c00f5603c85cfce48f0c439c7c90763f1e105

We installed a wsl copy of ubuntu as the quick workaround and initially the same error persisted - the docker-desktop instance was still the default wsl env. Setting ubuntu as the default resolved the problem.

Ah, I see in https://github.com/dorssel/usbipd-win/blob/30e508825eca1af023418683e93bbdc25d0948c6/Usbipd/Wsl.cs#L396 this is a known limitation.

This usage reports when it fails: https://github.com/dorssel/usbipd-win/blob/30e508825eca1af023418683e93bbdc25d0948c6/Usbipd/Wsl.cs#L524

Is there another usage of bash elsewhere? I noted the error says just `bash`, not `/bin/bash`

dorssel, 2025-04-02:
only the auto-attach script requires bash. without --auto-attach it should work, though.

andrewleech, 2025-04-02:
Ah good to know, yes I see that now and why it was showing just `bash`! 

I had an idea for reporting the error more clearly, how does this look? https://github.com/dorssel/usbipd-win/pull/1177