dorssel/usbipd-win #1183: Problem with UTN Manager attached dongles crashes usbipd

dockeratso, 2025-04-05:
This issue is related to #947 but causes **usbipd** to crash (see error log below).

In best case it would be nice to use such a dongle in WSL but if not possible then a better error message could be written for that dongle.

Version usbipd-win 4.0.0

C:\Users\scienta>usbipd list
Unhandled Exception: System.AggregateException: One or more errors occurred. (DEVPKEY_Device_LocationInfo returned 'SEH myUTN;CodeMeter-Stick', expected form 'Port_#0123.Hub_#4567')
 ---> System.NotSupportedException: DEVPKEY_Device_LocationInfo returned 'SEH myUTN;CodeMeter-Stick', expected form 'Port_#0123.Hub_#4567'
   at Usbipd.ConfigurationManager.GetBusId(UInt32) + 0x207
   at Usbipd.UsbDevice.GetAll() + 0xff
   at Usbipd.CommandHandlers.Usbipd.ICommandHandlers.List(Boolean, IConsole, CancellationToken) + 0x36
   at Usbipd.Program.<>c__DisplayClass11_4.<<Run>b__39>d.MoveNext() + 0xa7
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Invocation.AnonymousCommandHandler.<InvokeAsync>d__6.MoveNext() + 0x1ca
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext() + 0xe8
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass17_0.<<UseParseErrorReporting>b__0>d.MoveNext() + 0xed
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass12_0.<<UseHelp>b__0>d.MoveNext() + 0xbf
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseVersionOption>b__0>d.MoveNext() + 0x1da
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass19_0.<<UseTypoCorrections>b__0>d.MoveNext() + 0x13b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__18_0>d.MoveNext() + 0x193
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass16_0.<<UseParseDirective>b__0>d.MoveNext() + 0x101
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__5_0>d.MoveNext() + 0x194
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext() + 0x14d
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Invocation.InvocationPipeline.<<InvokeAsync>g__FullInvocationChainAsync|2_0>d.MoveNext() + 0xf7
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Parsing.ParseResultExtensions.<InvokeAsync>d__0.MoveNext() + 0xc8
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Parsing.ParserExtensions.<InvokeAsync>d__3.MoveNext() + 0xb5
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean) + 0x71
   at Usbipd.Program.Run(IConsole, ICommandHandlers, String[]) + 0x1eb1
   at usbipd!<BaseAddress>+0x3892f0

dorssel, 2025-04-05:
The real error is:

> Unhandled Exception: System.AggregateException: One or more errors occurred. (DEVPKEY_Device_LocationInfo returned 'SEH myUTN;CodeMeter-Stick', expected form 'Port_#0123.Hub_#4567')

This means you have your dongle attached to an incompatible hub. So, it's not the dongle, but the hub it is connected to that is the problem. Try connecting it to a different port (of a different hub).

The reason why the hub is incompatible is probably because it uses non-Microsoft driver. Can you locate the *hub* in the device manager and check which drivers are installed for it?

Besides that, of course usbipd-win shouldn't crash. I'll see if I can fix that.

dorssel, 2025-04-05:
Oh, and I see you are using a really old version 4.0.0. This was already fixed a long time ago. Won't make your hub work, but usbipd no longer crashes.

Old code is here, when the crash was still deliberate: https://github.com/dorssel/usbipd-win/blob/v4.0.0/Usbipd/ConfigurationManager.cs#L156

New code is here, with a mention about the incompatible hub issue: https://github.com/dorssel/usbipd-win/blob/v5.0.0/Usbipd/ConfigurationManager.cs#L159-L161

closed 2025-04-05 by dorssel

dockeratso, 2025-04-05:
> Closed #1183<https://github.com/dorssel/usbipd-win/issues/1183> as completed.
>
>
> Guaranteed Confidentiality
>
> You are hereby formally notified that all information contained in this communication and any attachments shall be deemed strictly confidential and privileged unless explicitly stated otherwise. Please note that your use of confidential information may be governed, and restricted, by a non-disclosure agreement. The information contained in this communication and any attachments is disclosed for the sole use of the intended recipient(s). If you are not the intended recipient, you are hereby formally notified that any unauthorized review, use, disclosure or distribution of this message is prohibited. Please notify the sender immediately by replying to this message and destroy all copies of this message and any attachments.

Hi.
Thanks. I  updated to the newer version 5.0.0 and now a see that the device is not compatible.

Skickat från min iPhone

dockeratso, 2025-04-05:
Hi.

The driver for the hub is the se sehutn.sys
which seems not to be compatible.

Mvh Anders

Skickat från min iPhone