dorssel/usbipd-win #1202: Please support com0com COM ports devices!

MrMarvel, 2025-05-17:
I want to share com ports to wsl. then to docker.
but when I want to list usb devices on host. There is no com0com device!

[image]

dorssel, 2025-05-18:
Besides the warnings (which are real!), you will need to specify which USB devices provides that COM functionality. `usbipd-win` does not forward COM ports; instead, it forwards USB devices (which in turn may very well provide COM ports).

dorssel, 2025-05-20:
You can share any USB device that is listed by `usbipd list`. My guess is that your virtual com port provider is *not* a USB device.

MrMarvel, 2025-05-20:
Yes com0com does not provide usb device.
Do you know any software for creating virtual usb device (maybe also working with com)?

closed 2025-05-21 by dorssel