dorssel/usbipd-win #1217: Server restarts when attaching USB device using usbip.exe (non-WSL Windows client)

T04N, 2025-06-11:
Hi, I'm encountering a serious issue when trying to use usbipd-win to share a USetup:
Server: Windows machine running usbipd (via dorssel/usbipd-win)

Client: Another Windows machine (not WSL), using usbip.exe to attach the shared USB device.

Problem:
Whenever I attempt to attach the USB device from the client using usbip.exe, the server machine immediately restarts without warning. It seems like a crash or forced reboot.

[image]

Command I used on the client:

usbip.exe attach --remote=<server-ip> --busid=<bus-id>

After running this command, the server ( PC )  reboots instantly.
Additional Info:
When I use [this prebuilt tool](https://gitlab.com/alelec/wsl-usb-gui) (WSL USB GUI) instead of usbip.exe, the USB attach works perfectly, and the server does not reboot.
This suggests that the issue may be related to how usbip.exe communicates with the server, or perhaps something in the usbipd handling of a non-WSL Windows client request.

Request:
Has anyone experienced similar behavior?

Is there a known issue when using usbip.exe from a regular Windows (non-WSL) environment?
Any idea what could cause the server to restart like this?

Thanks in advance for your help!SB device from my Windows machine.

dorssel, 2025-06-11:
You are saying that the machine running `usbipd-win` reboots when a remote client tries to attach? Is there a blue screen? Does the Windows EventLog say why it rebooted?

dorssel, 2025-06-11:
BTW, is the `usbip` client running on a different physical machine, or on a VM on the same physical machine as the server?